SpinalHDL/SpinalHDL #181: Add Number/Natural ranged type

typingArtist, 2019-01-03:
From a Gitter discussion in November:

I think that VHDL does it quite right with its ```integer``` type and ranged subtypes, except for two things:

1. the hassle to convert it back and forth from bit vectors, and
2. the inability to check (or even infer) the range of the result of an arithmetic operation.

Both issues could be approached easily by SpinalHDL if we introduce a ```Natural``` type, acting as a base type for both ```UInt``` and ```SInt```, that supports ```min``` and ```max``` value bounds instead of a width, while both ```min``` and ```max``` can be negative, zero or positive. The mathematically correct ```+``` and ```-``` operators should always return a result that can carry any output value, i.e. the bounds for the result of ```+``` are ```min = op1.min + op2.min```, ```max = op1.max + op2.max```. 

```+|``` and ```-|``` could be used to ask for a saturated result based on the bounds of the ```op1``` operand.

Finally ```+%``` and ```-%``` would wrap around in ```op1```'s bounds which would resemble more or less the function of the current ```+``` and ```-``` of ```UInt``` and ```SInt```. More or less, because it would really have to mathematically wrap around based on ```op1```’s ```min``` and ```max``` values which might be an expensive operation if the bounds are not aligned to powers of 2.

We could also start with the requirement that ```+%``` and ```-%``` only work on ```op1``` if ```op1.min  = 0``` and ```op1.max = 2^n-1```, or ```op1.min = -2^(n-1)``` and ```op1.max = 2^(n-1)-1```, i.e. fully compatible with the current ```UInt``` and ```SInt``` behavior.

_I have not added all comments from the Gitter chat here, feel free to add your thoughts._

Dolu1990, 2019-01-04:
I agree that there is currently a missing thing, especialy for DSP stuff.

What's about more general emulated data type that bring this min/max notion + fixed point position ?

Dolu1990, 2019-09-06:
<3

jijingg, 2019-09-07:
I still have some question about use `min`,`max` to creat Numric, it's totally ok in software.  
but in hadware, as you say , that will be expensive if bounds are not aligned to powers of 2.  
you can imange that :   
```scala
val x = Numeric(min=-40,max=100, 8 bits)
val y = Numeric(8 btis)
x :=y 
```
It in verilog will be 
```verilog
// Comparator equals a constant-Adder
// constant-Adder smaller than FullAdder
always @（*）
if(y<-40)      //one cAdder one Mux
    x = -40；
else if(y>100) //one cAdder one Mux
    x = 100;
else 
    x = x ;
```
the cost of  Numeric(min=-40,max=100, 8 bits) is **two 8bit cAdder and two Mux**.
and also cost for **timing** either.

so Numeric used in HW should be very carefully and the designer should clearly know what they do.
Otherwise, abuse of `min max` will lead area cost and bad-timing.

From another point of view， 8 bit Wire give capability -128 ~127, why Limited to -40 ~ 100， just use 8 bits, it totally safe for -40 ~ 100.  if -40 ~ 100 is algorithm feature , just Explicit us x > -40.

so most of time, if the value in -40 ~ 100 , creat Numeric(min=-128,max=127,8 bits)
if value in -40 ~ 50, creat Numeric(min=-64,max=63,7 bits)

## Integer

Numeric(min=-128,max=127,8 bits) Simplify is Numeric(8 bits)
Numeric(min=-64,  max=63,7 bits)   Simplify is Numeric(7 bits)
That's  exactly  SInt( 8 bits)  SInt(7 bits)

## Fraction 

example:
Numeric(min =- 1.34, max =73.8,  8 bits) will be fail , we need explicit address decimal postion.
we may creat like, 
Numeric(min =- 1.34, max =73.8, width= 8 bits,fraction = 3 bits) 
Here comes the problem.
infomation of max min  is confilict to width,
- if we give preference to width  

we got Numeric(max= ~16, max= 15.875, 8 bits ,fraction = 3 bits)
Simplify Numeric(width= 8bits , fraction = 3bits)
That's exactly  SFix(4 exp ,-3 exp)

- if we give preference to min, max 

we got Numeric(min =- 256, max =255.875, width = 11 bits, fraction = 3 bits)  
Simplify Numeric(width= 11bits , fraction = 3bits)  
That's exactly  SFix(7 exp ,-3 exp)  

## 
**Finally  they all come to SInt or SFix .**

Dolu1990, 2019-09-07:
> that will be expensive if bounds are not aligned to powers of 2. 
If the storage is done in a not aligned manner yes, but that wasn't my intend.

You can Have a Numeric Min = -30 max = 62. and having it stored in a 7 bits, where the value 0  is encoded in 0000000 binary.

About the Fraction section.
I agree Numeric(min =- 1.34, max =73.8, 8 bits)  isn't good.
There could be many different constructors, the thing which realy mather at the end is what is realy stored at elaboration time in the Numeric class. As far i can see : 
- Min / Max (I'm not exactly sure how we should store the Min/Max)
- Resolution (increment between each representable values)
- Number of bits

What's about having also Numeric(min =- 256, max =255.875, resolution = -3 exp) ?
Also i find 255.875 quite enoying
SNumeric(amplitude = 8 exp, resolution = -3  => 12 bits
UNumeric(amplitude = 8 exp, resolution = -3 exp) ? => 11 bits

jijingg, 2019-09-11:
@Dolu1990 Actually，I still don't have a clear understanding of Numeric's creation and where and how to use it.

From my previous experience. My steps are:  
1. Performance Simulation at Full Precision Floating Point
2. Performance Simulation at Fixed Precision 
3. RTL design with fixe-point width information which provided step2

so my dsptool are created for fixed-point simulation , after fixed-point simulation we can get accurate fixed width information which guide IC-engineer to design RTL .

i create [FixData](https://github.com/jijingg/SpinalHDL/blob/dsptool/lib/src/main/scala/spinal/lib/dsptool/FixData.scala)(in dsptool branch in my fork, Not PR yet),  
 the `signed, max, min, resolution` information are stored in [QFormat.](https://github.com/jijingg/SpinalHDL/blob/dsptool/lib/src/main/scala/spinal/lib/dsptool/QFormat.scala)

dsptool: https://github.com/jijingg/SpinalHDL/tree/dsptool/lib/src/main/scala/spinal/lib/dsptool

example:
```scala
val x = FixData(-3.785333,SQ(8,4))
=> FixData: -3.8125, Quantized by QFormat: Q(8,4,signed)
x.bin => 0b11000011
x.oct =>      0o103
x.hex =>       0xc3
```
Instead, we can look at the numerical value of a Bits value under a QFormat
```scala
toFixData(0xFFAE, SQ(8,4)) => FixData: -5.125, QFormat: Q(8,4,signed)  
toFixData(322111, SQ(8,4)) => FixData: -8.0,   QFormat: Q(8,4,signed)
toFixData(322111, UQ(8,4)) => FixData: 7.9375, QFormat: Q(8,4,unsigned)
toFixData(-322111,SQ(8,4)) => FixData: -8.0,   QFormat: Q(8,4,signed)
toFixData(-322111,UQ(8,4)) => FixData: 0,      QFormat: Q(8,4,unsigned)
toFixData(-0x0f,  SQ(8,4)) => FixData: -0.9375,QFormat: Q(8,4,signed)
```
dose it what you want ?

This has basically solved my requirements in the design of communication algorithms.

Maybe in the future, step 2 and step 3 will be merged into a super Dsptool, but I don't want to do that yet.

Dolu1990, 2019-09-11:
Ahhh so, i get it now (your flow), So basicaly, in ALU design, there is no step 1/2. the inputs of the ALU are specified by the register file and literals width, but then the range of things can increase along the pipeline until it endup into some saturation/truncation units and written back into the register file.

To resume : 
1) inputs of a specified size
2) math which never drop informations and expend width on the go
3) sat/truc
4) outputs of a specified size

> dose it what you want ?
I'm not sure about the question, sorry :)

jijingg, 2019-09-12:
in ALU design ，According the resume you described. 

the problem may more simpler. 

only use  `+^` `-^` can achive it, dose it ?  
other operation like `*` are already safe.    
if bit width increase exceed of set value , just use `sat` .
No more 

the seconde question is :   
I just want to ask , is FixData by Qformat your expectation of Numeric?

Dolu1990, 2019-09-14:
I think '8 bits +^ (4 bits +^ 4 bits)'  and '8 bits +^ 4 bits + 4 bit' are bad solution (compared to Numeric), because it mean you had to manualy be extra carefull with where to place the +^ and where the + to get the behaviour you wanted.
The day the bits width of things change you have to recheck and sometime fix all your pipeline.
With numeric there would be no sort of things like that. Just '+' operator, and it would be good forever <3

The idea is to avoid having to enginner the width of each every little bit of the aritmetic, but stay more generic and only having to specify truncations and saturations.

jijingg, 2019-09-15:
Yes, at RTL level engineer need manually handle this .
What Numeric does seems like the HLS things( just my personal feel, may not right).
To be honestly, I still don't have a clear idea of what will Numeric look like :laughing:
If there is a real example(Pseudo-code is also ok), it may be more telling.

typingArtist, 2019-09-16:
with Numeric, you get something like the following. The initial implementation will be for integers, but fixed-point will follow.

```scala
  val a = Reg(Numeric(min=0, max=127)) // implemented as a UInt(7 bits)
  val b = Reg(Numeric(min=-16, max=15)) // implemented as a SInt(5 bits)
  val c = Reg(Numeric(min=-1, max=3)) // implemented as a SInt(3 bits)
  val d = a + b - c // d is Numeric(min=-19, max=143) // implemented as a SInt(9 bits)
  val e = b * c // e is Numeric(min=-48, max=45) // implemented as a SInt(7 bits)
  a := a + 1 // this fails because RHS is Numeric(min=1, max=128) which
             // doesn’t fit the LHS
  e := b + c * 2 // this succeeds. RHS is Numeric(min=-18, max=21)
                 // which fits the LHS 
  c := c +% 1 // this succeeds. RHS will overflow from 3 to -1, SpinalHDL will
              // infer overflow logic for handling that trick
  a := a +% 1 // this succeeds. RHS keeps the range of a and allows
              // overflow/modulo addition. SpinalHDL is clever enough to
              // see that inferral of overflow logic is not needed because a
              // takes its full implementation range of 7 bits for min..max
  c := c +% 13 // this succeeds. Since the range of c is 5 steps, this
               // is equivalent to c := c +% 3 and will infer proper overflow logic
  c := c +% b // this succeeds but requires quite some modulo logic
  b := b +% c // this succeeds without added modulo logic, since
              // b takes its full Numeric range.
  c := b.cast(c) // this succeeds. b is an unsafe hard cast to the range
                 // of c. Any valid value of b in the range of c will be
                 // represented correctly, any other value will lead to
                 // undefined results
  c := (c + b).cast(c) // this succeeds. RHS addition result is hard cast
              // unsafely to the physical width of c. No overhead logic is
              // implemented. Overflow is undefined, for both
              // simulation and implementation. For example, most
              // likely c===3 and b===9 will lead to a still invalid c===-4.
  a := a +| b // this implements a saturated addition. It is equivalent to
  a := mux (a + b > a.max, a.max, 
                 mux(a + b < a.min, a.min, (a + b).cast(a)))
  a := a +| 1 // this implements a saturated addition. Because of the
              // constant, SpinalHDL can optimize it to
  a := mux(a === a.max, a.max, (a + 1).cast(a))
```

More to come.

jijingg, 2019-09-23:
@typingArtist Thanks for your detailed examples.
I have a rough idea of where Numeric is used.  In Numeric, the fix operation(round, saturation) is auto completed, and the compiler guarantees the operation is safe. 

For some scenarios(HLS, develope speed concern ), it's really convenient. 
but for low-cost, timing-sensitive design, it may be a little expensive,   
So I personally prefer that all three levels are reserved for users,  rather than forced unification to one high-level . User choose according to their needs. 

[image]

Dolu1990, 2019-09-23:
@jijingg In which case Numeric has higher cost or worst timings than FixPoint/UInt/SInt ? 
Basicaly, everything you can express with FixPoint/UInt/SInt should be expressable with Numeric, isn't it ?

jijingg, 2019-09-23:
Yes, i'm wrong, Numeric is no overhead, it's just a high-wrap of fixPoint.

after calculate, if the Intermediate node value is `(-12,100)`,you will just fixTo SInt(8 bits), not do `-12<SInt(8 bits)<100` operation.

`min, max` only affects bit width , not actually operate on hardware. right?