SpinalHDL/SpinalHDL #193: .clearAll and .setAll don't work anymore for UInt

tomverbeure, 2019-01-19:
I some older code, I use .clearAll and .setAll on UInt.

After updating to the latest SpinalHDL version, this code errors out like this:

```
[info] [Progress] at 0.587 : Elaborate components
[error] Exception in thread "main" java.lang.NoSuchMethodError: spinal.core.UInt.setAll()V
[error] 	at math.FpxxAdd$$anonfun$11.apply$mcV$sp(FpxxAdd.scala:227)
[error] 	at spinal.core.when$.apply(when.scala:83)
```

Here's some of the code on which it fails:
https://github.com/tomverbeure/math/blob/2d9fbf27218d7574083fee5c417021c707ce4d8c/src/main/scala/math/FpxxAdd.scala#L43-L49

Tom

tomverbeure, 2019-01-19:
I assume that this failure has something to do with this change:
https://github.com/SpinalHDL/SpinalHDL/commit/cced8fa62d5483a3a88620253103551d9c68518f

Dolu1990, 2019-01-19:
That's weird, i just tried 

```scala
    SpinalVerilog( new Component {
      val output = out(UInt(15 bits))
      output.setAll
    })
```

And that's fine.

Have you done a sbt publishLocal on the SpinalHDL repository by the past ?

tomverbeure, 2019-01-19:
No, I’ve never used publishLocal before.

Tom

Dolu1990, 2019-01-19:
I just cloned the master branch of the repo, updated 

```scala

lazy val root = (project in file(".")).
  settings(
    inThisBuild(List(
      organization := "com.github.spinalhdl",
      scalaVersion := "2.11.6",
      version      := "0.1.0-SNAPSHOT"
    )),
    name := "Fpxx"
  ).dependsOn(vexRiscv)

lazy val vexRiscv = RootProject(uri("git://github.com/SpinalHDL/VexRiscv.git#b5caca54cd97e968f8b6c7494aa8aed7388a23c6"))

//If you want a specific git commit : 
//lazy val vexRiscv = RootProject(uri("git://github.com/SpinalHDL/VexRiscv.git#commitHash"))

//If the dependancy is localy on your computer : 
//lazy val vexRiscv = RootProject(file("local/path/to/the/VexRiscv/sbt/project/VexRiscv"))

fork := true
```

run math.TopVerilog, and it was fine
Have you tried a sbt clean ?

tomverbeure, 2019-01-19:
After `sbt clean`, I get the following:
```
[warn] Found version conflict(s) in library dependencies; some are suspected to be binary incompatible:
[warn]
[warn] 	* com.github.spinalhdl:spinalhdl-lib_2.11:1.3.0 is selected over 1.2.0
[warn] 	    +- com.github.spinalhdl:superproject_2.11:1.0.0       (depends on 1.3.0)
[warn] 	    +- com.github.spinalhdl:vexriscv_2.11:1.0.0           (depends on 1.2.0)
[warn]
[warn] 	* com.github.spinalhdl:spinalhdl-core_2.11:1.3.0 is selected over 1.2.0
[warn] 	    +- com.github.spinalhdl:superproject_2.11:1.0.0       (depends on 1.3.0)
[warn] 	    +- com.github.spinalhdl:spinalhdl-lib_2.11:1.3.0      (depends on 1.3.0)
[warn] 	    +- com.github.spinalhdl:vexriscv_2.11:1.0.0           (depends on 1.2.0)
[warn]
[warn] Run 'evicted' to see detailed eviction warnings```

So that's probably a clue.

I'm running this from my `rt` project, which included `math`.

Digging deeper...

Tom

tomverbeure, 2019-01-19:
Ok. I upgraded my `rt` projects to the latest SpinalHDL version, but `math` was an external library that was not upgraded to the latest version.

After upgrading that one as well, it's working.

Speaking about upgrading SpinalHDL version: it spent 3 hours last night trying to figure out how to do that. The final clue was that no only you need to update `build.sbt` but also files in `./project`. And even then, of course, did `rt` not compile.

I've added an `Using a Local SpinalHDL Version` chapter to my [SpinalHDL Snippets](https://tomverbeure.github.io/rtl/2018/10/14/SpinalHDL-Snippets.html), but ideally, this kind of stuff should be documented somewhere in the official documentation?

Tom

Dolu1990, 2019-01-20:
One mistake in https://github.com/SpinalHDL/SpinalTemplateSbtDependencies/blob/superproject/build.sbt, is that the SpinalHDL version is currently not specified, but come from the VexRiscv depedancy, i will modify that.

About 

> Speaking about upgrading SpinalHDL version: it spent 3 hours last night trying to figure out how to do that. The final clue was that no only you need to update build.sbt but also files in ./project. And even then, of course, did rt not compile.

I figured out that using the sbt lazy val vexRiscv = RootProject(uri("git://github.com/SpinalHDL/VexRiscv.git")) isn't safe, as it can keep quite hold version of the git, and will not try to update it, until you flush the ~/.sbt cache.

So i updated the https://github.com/SpinalHDL/SpinalTemplateSbtDependencies/blob/superproject/build.sbt to use a git module and import it with a relative path instead.

> I've added an Using a Local SpinalHDL Version chapter to my SpinalHDL Snippets, but ideally, this kind of stuff should be documented somewhere in the official documentation?

It is already documented, but the doc is too big to find it :/
https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-commited-on-git

I need anyway to fix the publish-local into publishLocal ^^

closed 2019-01-29 by Dolu1990