SpinalHDL/SpinalHDL #196: Build break with latest dev commit

tomverbeure, 2019-01-22:
After updating my local SpinalHD dev to the latest commit (https://github.com/SpinalHDL/SpinalHDL/commit/c3b555afb3914427ac268f2294aeee77cd174d88), I get the following error when doing a clean and then a compile:

```
[error] Symbol 'type spinal.core.BaseTypeFactory' is missing from the classpath.
[error] This symbol is required by 'package spinal.core.package'.
[error] Make sure that type BaseTypeFactory is in your classpath and check for conflicting dependencies with `-Ylog-classpath`.
[error] A full rebuild may help if 'package.class' was compiled against an incompatible version of spinal.core.
[error] Symbol 'type spinal.core.BaseTypeCast' is missing from the classpath.
[error] This symbol is required by 'package spinal.core.package'.
[error] Make sure that type BaseTypeCast is in your classpath and check for conflicting dependencies with `-Ylog-classpath`.
[error] A full rebuild may help if 'package.class' was compiled against an incompatible version of spinal.core.
[error] two errors found
[error] (ProjectRef(uri("file:/home/ubuntu/projects/SpinalHDL/"), "core") / Compile / compileIncremental) Compilation failed
[error] Total time: 31 s, completed Jan 22, 2019 2:05:04 AM
```

When reverting back to the previous commit (https://github.com/SpinalHDL/SpinalHDL/commit/6641ef48265291bd9031e50baa7ecbf5c9609b25), it works fine again.

Tom

Dolu1990, 2019-01-22:
have you try  a "sbt clean" ?
The [error] Symbol xxx is missing from the classpath. is likely a incrémental compilation bug, i often have it when modifying things in the spinal.,core

tomverbeure, 2019-01-22:
Absolutely! I do that religiously these days. :-)

Tom

Dolu1990, 2019-01-22:
The issue i have is the travis is currently ok (https://travis-ci.org/SpinalHDL/SpinalHDL/builds)
I have to check, but now it is sleeping time :)

tomverbeure, 2019-01-22:
After going through another recompile round, I also had the issue with earlier version.

So I tried deleting the full ~/.sbt directory, ./target directories etc etc.

Eventually, I tried the `publishLocal` route (which I had avoided until now: I prefer linking to a directory instead of overwriting the cache), and after that it works.

sbt is a little bit of a cancer.

Anyway, some kind of state is probably still lingering somewhere, but it's working now. Closing.

closed 2019-01-22 by tomverbeure

Dolu1990, 2019-01-22:
Did you tired to do the sbt clean into the SpinalHDL project itself instead of the sub project ?

Sure, sbt can lead to much troubles, there is definitvely some features to avoid ^^