SpinalHDL/SpinalHDL #202: Add comparison with Chisel to the documentation

JobNoorman, 2019-02-26:
Since SpinalHDL seems very similar to Chisel, I think it would be useful to provide a detailed comparison. The only thing I can find is an archived [Reddit thread](https://www.reddit.com/r/chisel/comments/4ivevd/spinalhdl_a_chisel_fork/) that's almost three years old that lists some differences. It also contains a [dead link](http://spinalhdl.github.io/SpinalDoc/#what-are-the-differences-between-chisel-vs-spinal-) to a SpinalHDL documentation page. Since this thread was started before the release of Chisel3, I don't know how many of its points are still valid.

I think it's very important to have this comparison because Chisel seems to be much wider used and supported. Choosing SpinalHDL over Chisel for a new project therefore seems risky and only justifiable if there are some very convincing reasons to prefer it.

Dolu1990, 2019-02-26:
I agree SpinalHDL is realy similar to Chisel.
I removed the comparison from the dead link for multiple reasons : 
- Kindness for Chisel peoples
- Chisel 3 progresses
- I'm not award about all the Chisel 3 changes
- That page was made in the time of Chisel 2, when Chisel 2 had so many issue and i had to explain why i "forked" with SpinalHDL.

Today, i would say that SpinalHDL key points are : 
- Strongly typed (especialy bitwidth of signals)
- A bit more meta, as you have access to the whole netlist AST durring the Scala elaboration, so you can edit and rework the netlist. As far i know, with Chisel 3, you have to switch to FIRRTL to do that.
- ClockDomain support, especialy for FPGA stuff, but Chisel guys recently made some progress on that.
- The spinal.lib

About Chisel being wider used and supported, that's right ^^
**troll on** Anyway, nobody is using Chisel nor SpinalHDL in the industry, and if you do, you should be excommunicate from the holly EDA world right now **troll off**
More seriously, that's all about threshold levels.  You could say the same for Chisel vs SystemVerilog, but at the end it do not matter. Which matter is that for all serious bugs in SpinalHDL, a fix was produced in the few days. So commitment is there.
Also, once the language is stable and allow you to do what ever you want, you potentialy could leave it as it is for over 9000 years. And SpinalHDL is quite stable.
I don't realy see any risk into using SpinalHDL over Chisel.

About SpinalHDL users, we mainly talk there : 
https://gitter.im/SpinalHDL/SpinalHDL

closed 2019-04-24 by Dolu1990

jijingg, 2019-08-05:
I've been learning to use Chisel for a while, and recently decided to full use SpinalHDL in my IC design work. In fact, when I studied Chisel, I already knew SpinalHDL exists. At that time, I didn't take it seriously and didn't even bother to look at it. Many people may think the same way as me. Why do I spend time learning the same thing again. The idea made me make a big mistake. Later, when I got a chance to re-read his documents and source code, I was shocked. That's exactly what I wanted. That's the right way to open CHISEL(Constructing Hardware In Scala Embedded Language).

SpinalHDL design is more in line with IC hardware developers'habits. If I am not mistaken, the author must have IC development background, clear module division and conceptual design in line with hardware development. The author konw what to do and what not to do. But Rocket-chip is messy and lacks documentation, and based on the first version of Chisel, there are too many historical burdens. On the contrary, **pinsec** is so simple, clear and easy to read. There are also beautiful designs such as Stream Flow edc. and very useful concepts such as **ms, KiB, Mhz**, which will be more efficient to provide parameterization capability. I don't think what Chisel can do but SpinalHDL can not do, in nature they all based on Scala. And chisel-iotesters is a completely wrong direction, Believe me, don't have any IC engineer will use it.

Chisel's development team comes from school. They are creative, but  lack engineering experience. There are great differences between design and engineering. I don't totally deny Chisel here. At least chisel has made SCALA-based hardware development popular and made more people Know there is an new and efficient hardware design method exists. But from a product point of view, SpianlHDL is a better choice. Whether you decide to continue using Chisel or SpinalHDL in the future, It's worth reading the following SpianalHDL documentation. 

Thank you most sincerely to Charles Papon and other contributors.

Dolu1990, 2019-08-05:
> Thank you most sincerely to Charles Papon and other contributors.

Thanks :)

BracketMaster, 2022-10-06:
I worked on a fairly sizable project in Chisel3 in 2021 before coming to SpinalHDL. I remember Chisel causing me lots of trouble - particularly with poking vectors in Simulation, casting bits, and printing enums in simulation.

Chisel also had a number of other minor quirks that quickly added up and made the experience of writing Chisel rather unbearable.