SpinalHDL/SpinalHDL #205: Make Vec[BitVector] single bit assignable

Dolu1990, 2019-03-27:


numero-744, 2022-06-05:
Isn't `myVec.asBits(i)` okay?

Dolu1990, 2022-06-06:
It isn't because asBits will return a copy of the myVec value, which when assigned, will affect the copy, and not the original Vec.

But that issue is actualy already fixed ^^

```scala
    val vec = Vec.fill(4)(Reg(UInt(8 bits)) init(0))
    vec(in UInt(2 bits))(3) := in(Bool())
```

```verilog
  reg        [7:0]    vec_0;
  reg        [7:0]    vec_1;
  reg        [7:0]    vec_2;
  reg        [7:0]    vec_3;
  wire       [3:0]    _zz_2;

  assign _zz_2 = ({3'd0,1'b1} <<< _zz_1);
  always @(posedge clk or posedge reset) begin
    if(reset) begin
      vec_0 <= 8'h0;
      vec_1 <= 8'h0;
      vec_2 <= 8'h0;
      vec_3 <= 8'h0;
    end else begin
      if(_zz_2[0]) begin
        vec_0[3] <= _zz_vec_0;
      end
      if(_zz_2[1]) begin
        vec_1[3] <= _zz_vec_0;
      end
      if(_zz_2[2]) begin
        vec_2[3] <= _zz_vec_0;
      end
      if(_zz_2[3]) begin
        vec_3[3] <= _zz_vec_0;
      end
    end
  end
```

closed 2022-06-06 by Dolu1990