SpinalHDL/SpinalHDL #206: asBits circumvents width mismatch check

JobNoorman, 2019-04-10:
The following code correctly issues a width mismatch error:
```scala
val u1 = U(0, 1 bits)
val u2 = U(0, 2 bits)
u1 := u2
```
However, when casting the values to `Bits` first, the error is gone:
```scala
u1.asBits := u2.asBits // No error here
```
Interestingly, the bit width is correctly preserved by the cast:
```scala
println(s"${u1.asBits.getBitsWidth} ${u2.asBits.getBitsWidth}") // Prints "1 2"
```
Also, a width mismatch *is* reported when creating `Bits` directly:
```scala
val b1 = B(0, 1 bits)
val b2 = B(0, 2 bits)
b1 := b2 // Error correctly reported here
```
So something seems to go wrong when casting to `Bits` using `asBits`.

JobNoorman, 2019-04-10:
After some further experimentation, it seems to be that the error is gone when the _left hand side_ of an assignment is casted to `Bits`, when only the right hand side is casted the error is still reported:
```scala
val b1 = B(0, 1 bits)
val u2 = U(0, 2 bits)
b1 := u2.asBits // Error correctly reported here
u2.asBits := b1 // No error here
```

Dolu1990, 2019-04-10:
Hi,

u2.asBits := b1 will not assign u2, but a Bits wire which is assigned by u2.
Do you see the difference i mean ?

u2.asBits := b1  is basicaly useless.
 
u2 := b1.asUInt.resized  is fine
u2 := b1.asUInt(2 bits)  is fine too, if the syntax existe, i'm not sure , i have to check

JobNoorman, 2019-04-10:
> u2.asBits := b1 will not assign u2, but a Bits wire which is assigned by u2.
> Do you see the difference i mean ?

I see now, I was a bit afraid that what I tried to do didn't make sense. It is a bit counter-intuitive, though, since the documentation describes all the `asX` functions as "binary casts" which would imply being able to assign to the result of such a cast. However, the actual implementation of these function is no cast at all.

> u2 := b1.asUInt.resized is fine

Sure but in my case I actually _wanted_ to get a width mismatch error while a didn't get one that I expected.

Maybe I should explain my actual problem. I have a function that has two values of type `Data` (or `BaseType`) and needs to assign one to the other. I don't actually care about the type of both values (one could be `Bool` the other `UInt`) as long as their bit width is the same. A regular assignment doesn't work if the types are not the same so I have to do some kind of cast. Unfortunately, all types of cast I tried either don't work (like in my original post) or convert bit widths as well:
```scala
def foo(v1: Data, v2: Data): Unit = {
  v1 := v2 // Error if, e.g., v1=Bool() and v2=Bits(1 bits)
  v1.asBits := v2.asBits // No error but doesn't work as expected
  v1 := v2.as(v1) // Silently converts v2 to the same bit width as v1
  v1.assignFromBits(v2.asBits) // Silently converts v2 to the same bit width as v1
}
```
So is there a way to safely assign from one type to another? That is, while still checking bit widths?

Dolu1990, 2019-04-16:
Sorry, I forgot that issue XD

So, i tried your example, and in my case, 
x.assignFromBits(y.asBits) do not silently convert the width.

I can't reproduce that behaviour, can you provide me an full example ? including the main ?
Maybe there is something else under the hood :)

Dolu1990, 2019-05-08:
About   toto.asBits := miaou

Like this, right, it do not make sense. But imagine the following : 

```scala
val wuf = toto.asBits
when(something === 42){
   wuf := 666
}
```
This isn't exactly the same case, but not so far.

And this kind of case can happen even without you knowing : 
```scala
def func(a : UInt) = ... ... return a.asBits
...
...
val wuf = func(toto)
when(something === 42){
   wuf := 666
}
```

d1.assignFromBits(d2.asBits) // No error here

Ahh nice catch :D
The error is from here : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/Bool.scala#L163
It do not do any check, just task the LSB bit.

Do you wanna try to correct it with a pullrequest ?
Or i can do it, that's fine for me. as you want.
Basicaly just adding an assert on the width of the input signal would do the job.

Dolu1990, 2019-05-08:
Why my  last message appear before you ones XD ??

commit ccef1da5 (2019-05-09): Fix #206, Add width assertion on Bool.assignFromBits
Sorry, had to stress the thing to do a release XD
closed 2019-05-09 by Dolu1990