SpinalHDL/SpinalHDL #210: Add 1 bit to return type of Operator.UInt/SInt.Add/Sub?

jjf-oculus, 2019-04-26:
Generated logic (VHDL + Verilog) for addition and subtraction loses the high bit of the result. 

Taking an example from http://spinalhdl.github.io/SpinalDoc/spinal/core/components_hierarchy/:
```
class MyAdder(width: BitCount) extends Component {
  val io = new Bundle{
    val a,b    = in UInt(width)
    val result = out UInt(width)
  }
  io.result := io.a + io.b
}
```
new MyAdder(8 bits)
generates:
```
module MyAdder (
      input  [7:0] io_a,
      input  [7:0] io_b,
      output [7:0] io_result);
  assign io_result = (io_a + io_b);
endmodule

in verilog and in vhdl:

entity MyAdder is
  port(
    io_a : in unsigned(7 downto 0);
    io_b : in unsigned(7 downto 0);
    io_result : out unsigned(7 downto 0)
  );
end MyAdder;

architecture arch of MyAdder is

begin
  io_result <= (io_a + io_b);
end arch;
```

Which if (io_a + io_b ) > 255 produces an incorrect result.  Generating the correct logic requires manually resizing operations *prior* to the operation:

class MyAdder2(width: Int) extends Component {
  val io = new Bundle{
    val a,b    = in UInt(width bits)
    val result = out UInt(width + 1 bits)
  }
  io.result := io.a.resize(width + 1 bits) + io.b.resize(width + 1 bits)
}

(produces)
```
module MyAdder2 (
      input  [7:0] io_a,
      input  [7:0] io_b,
      output [8:0] io_result);
  wire [8:0] _zz_1_;
  wire [8:0] _zz_2_;
  assign _zz_1_ = {1'd0, io_a};
  assign _zz_2_ = {1'd0, io_b};
  assign io_result = (_zz_1_ + _zz_2_);
endmodule
```

wifasoi, 2019-04-26:
I think for the "base type" keep a strict size check is best for the long run.
There are similar Idea (implement a new type/wrapper) in this #181, but I don't know if someone is actively working on it (right now I mean).

Dolu1990, 2019-04-29:
I agree with @wifasoi 

Also, we would probably need a proper type for DSP stuff / conservative aritmetic.
As far i know nobody is one it at the moment

Dolu1990, 2019-06-15:
Realy need a DSP / aritmetic type

closed 2019-06-15 by Dolu1990

jijingg, 2019-08-07:
hi @Dolu1990.
i know that the DSP or MCU  without bits expand, they just alow highest bit spillover

But In some algrithm do not allow spillover, if it is really necessary to reduce the bit width, will be achieved through saturation or truncation. For example , In the field of communication hardware design, bit extension of addition and subtraction is very common. 

In order to facilitate the derivation of compiler bit width, I hope we can add 2 methods called "+!"-!"To express the need to expand 1 bit, without affecting the normal"+"and"-"operations.

Let users choose for themselves. If they need to expand, they can use "+!" "-!" 
```
  subResult := a -! b //means expand 1bit
  addResult := a +! b //means expand 1bit
```
Otherwise, use "+""-"
```
  subResult := a - b //means without bit expand 
  addResult := a + b //means without bit expand 
```

Dolu1990, 2019-08-07:
Hi @jijingg 

Having a dedicated operator for me is fine, basicaly, the main issue was to finding a nice symbolic operator without falling in the scala issue of having to many specific custom operators.

An alternative would be to have a fast way to extend a signal by one bit.
a.expanded + b.expended  
But i think the way with the +! -! operator is the way to go.

What about +@  and -@ ? 

@@ behing already use to concatain two aritmetic signals and preservating the type.

The issue i have with the '!' is it is already use as a 'not' operator in other cases.

jijingg, 2019-08-07:
@Dolu1990  
In my opinion , "!" means of Emphasizing, and looks like Number "1", "+!" call attention to users.  
another choice is "+&" which maybe friendly to users who come from chisel, "+&" is already used in chisel.  "+@" is good choice too ,The final verdict is in your hands :smile:  
but a.expanded  seems verbose, i don't recommended

Dolu1990, 2019-08-07:
@wifasoi @typingArtist what is your opinion ?