SpinalHDL/SpinalHDL #213: AxiLite4SlaveFactory.onReadPrimitive does nothing when haltSensitive=false

kaofishy, 2019-05-16:
When using `AxiLite4SlaveFactory.onReadPrimitive` with `haltSensitive` set to false, the associated anonymous function is never called.

I compared `AxiLite4SlaveFactory` with other `BusSlaveFactory` implementations and it looks like the difference is in the `build` method, `AxiLite4SlaveFactory` calls `doMappedReadElements` with `askRead = writeJoinEvent.valid` for some reason.

kaofishy, 2019-05-18:
Incidentally, as a side effect this means `multiCycleRead` doesn't do anything in `AxiLite4SlaveFactory` since it relies on `onReadPrimitive` to halt the bus. See: https://github.com/SpinalHDL/SpinalHDL/blob/8109f9ef9d9f17f9f264477fb29f991f4f641c96/lib/src/main/scala/spinal/lib/bus/misc/BusSlaveFactory.scala#L537-L549

Dolu1990, 2019-05-18:
Ahhhh askRead = writeJoinEvent.valid,   nice catch XD

I just submited a fix 0a8a9bfed86a144c4696f72765380294d1a5d277
Let's me know if you still have issues :)

kaofishy, 2019-05-19:
Yes this works perfectly now. Thanks!

closed 2019-05-19 by kaofishy

Dolu1990, 2019-05-19:
It is now released in 1.3.5 :)