SpinalHDL/SpinalHDL #214: Black boxes and bidirectional signals

ZiCog, 2019-05-17:
I wanted to create a Spinal black box for a simple SDRAM controller. The issue turns out to be that I can't connect to the controller's bidirectional data bus to the outside world from inside a black box.

I had found somewhere that Spinal has such things as "inout" signals and making connections with "<>" so I experimented a bit. I did read that Spinal did not support bidirectional (tristate) signals but seeing those things I had to experiment a bit. With no luck.

Is there a way to "tunnel" bidirectional signals out of a black box by some devious means? Or was I wasting my time? I could not find any mention of "inout" or "<>" in the documentation so I guess the latter.

What I really wanted to do was have my entire design in Spinal from the top level down. I now have a Verilog top level simply to contain the SDRAM controller and connect it to the FPGA pins with my Spinal project beside it.

The alternative would have been to modify the SDRAM controllers Verilog to provide in an out buses and a direction signal. I really did not want to do that. 

This is not a big deal. Just not aesthetically pleasing.

Please excuse this neophyte. Still very new to all things FPGA, Verilog and Spinal.

Oh, and thanks for this great piece of work. It makes my project a lot more fun.

Dolu1990, 2019-05-18:
Hi,

You can do that with blackboxes. In the SpinalHDL regression tests, i have a testcase which check it : 

https://github.com/SpinalHDL/SpinalHDL/blob/52be3be4ba2f243873d37b26b03fc998b3b3edf4/tester/src/test/scala/spinal/tester/scalatest/InOutTester.scala#L56

This single of the bus will be as a verilog inout signal : 
https://github.com/SpinalHDL/SpinalHDL/blob/52be3be4ba2f243873d37b26b03fc998b3b3edf4/tester/src/test/scala/spinal/tester/scalatest/InOutTester.scala#L31

And the connection from the black box to the toplevel io is done that way : 
https://github.com/SpinalHDL/SpinalHDL/blob/52be3be4ba2f243873d37b26b03fc998b3b3edf4/tester/src/test/scala/spinal/tester/scalatest/InOutTester.scala#L90

It is what you need, right ?

ZiCog, 2019-05-18:
Wow, sounds like I was pretty close to getting it right. Luckily I did not throw away my experimental code.

I had extended BlackBox and used "inout" on the data bus. Also noticed in came out as "inout" in the Verilog.  [https://github.com/ZiCog/sodor-spinal/blob/master/src/main/scala/sodor/sdram_controller.scala#L37](url)

I made the connections with ":="

I think my problem was that I used Tristate instead of Analog on my bus. Somehow I missed the section on Analog in the documentation. Which looks just like what I wanted!

I now have a working test of the SDRAM on a DE0-nano board in Spinal as an "almost" top level under a thin Verilog top level. It should be straight forward to put the sdram-controller back in black box and remove the Verilog. [https://github.com/ZiCog/sodor-spinal/blob/master/src/main/scala/sodor/sdram_controller_tb.scala](url)

Thank you hugely.

Dolu1990, 2019-05-18:
Ahh if you are interrested by it, there is a sdram in the spinalhdl lib :)

Have fun !

ZiCog, 2019-05-18:
It works!

My SDRAM test in Scala has now become my top level component using a verilog  SDRAM controller through a BlackBox. Both of which use  "...  = inout (Analog(Bits (16 bits)))" to define the data bus.
https://github.com/ZiCog/sodor-spinal/blob/master/src/main/scala/sodor/sdram_controller_tb.scala

Thanks for the heads up on the Spinal lib SDRAM controller.

I'll be sure to look into at some point. Just now I'm still getting familiar with FPGA, Verilog, Spinal, Scala. Not to mention processor design! As an exercise I'm trying to write everything I need for myself and don't want to put anything in there I don't understand to some extent. Just now I have no idea about AXI or Wishbone buses for example.

Thanks again for the help.

closed 2019-05-27 by Dolu1990