SpinalHDL/SpinalHDL #219: Conditional state transition for statemachine

rpls, 2019-06-03:
Is it possible to only transition a StateMachine when a condition is met? Obviously, this is possible with `when()` inside the states, but what if the condition is always the same? Let's say we have the following code:

```scala
val cond : Bool = /* ... */

val fsm = new StateMachine {
  val STATE1 = new State
  val STATE2 = new State
  /* ... */
  STATE1.whenIsActive {
    when(cond) {
      goto(STATE2)
     }
  }
  STATE2.whenIsActive {
    when(cond) {
      goto(STATEN)
     }
  }
}
```

Instead maybe something like this:

```scala
val cond : Bool = /* ... */

val fsm = new StateMachine {
  val STATE1 = new State
  val STATE2 = new State
  /* ... */
  /* Imaginary API */
  onlyTransitionWhen(cond)
  STATE1.whenIsActive {
    goto(STATE2)
  }
  STATE2.whenIsActive {
    goto(STATEN)
  }
}
```

I.e. only latch `stateNext` into `stateReg` when `cond` is `True`. Or would this be bad-practice?

rpls, 2019-06-03:
I guess without a new API, the following would achieve this?

```scala
val cond : Bool = /* ... */

val fsm = new StateMachine {
  val STATE1 = new State
  val STATE2 = new State
  /* ... */
  val _stateNext = enumDefinition()
  always {
    when(cond) {
      stateNext := _stateNext
    }
  }
  STATE1.whenIsActive {
    _stateNext := enumOf(STATE2)
  }
  STATE2.whenIsActive {
    _stateNext := enumOf(STATEN)
  }
}
```

Dolu1990, 2019-06-04:
hmm you want something behaving like a clock enable ?
Maybe using the clockEnable of ClockDomain would be good for your usecase ?

Currently, there is nothing in the FSM API itself.

rpls, 2019-06-04:
Yes, clockEnable might be an Idea. For reference, I‘m trying to get more
into SpinalHDL (and HDL in general) by building a CAN FD transceiver
component. So I‘m currently working on the FSM which sends and/or parses
the CAN frame, and the FSM pretty much only needs to react when a bit is
sampled, which may happen in time intervals anywhere from a few clocks to
hundreds. So I want to avoid something along the lines of
`when(input.valid)` in every state, both to clean up the code and to avoid
all the additional logic (unless I misunderstand something about what kind
of logic will be generated by this?).

So how would I go about this? Place the FSM in a ClockDomain with the same
Clock but with the clockEnable set when a bit is sampled? Might just be a
problem because the FSM will push some bits into a shift-register to send
out.

Dolu1990, 2019-06-04:
hmm
I understand that need. So maybe it would be great to add that to the FSM API. a global state transition enable. It would not be hard to do.

Do you wanna try to add this in the FSM API ?

rpls, 2019-06-04:
Yeah, I’ll fork the repo and send a PR.

Dolu1990, 2019-06-04:
Cool :D

PR #220 (2019-07-20): Add state machine variant with conditional state transition
So this is the current WIP for the state machine with conditional transition. I opted for an auxiliary constructor to pass the condition to the state machine, as to make it clear that this will change the behaviour of the fsm. The change works by introducing a "candidate" for `stateNext` named `stateNextCand`. Then `stateNext` will only be assigned this candidate, if the transition condition is met. I chose this design to simplify the treatment of the `onEntry`/`onExitTasks` and to avoid a giant `when()` around these.

I made the synt [... cut]
closed 2019-07-20 by Dolu1990