SpinalHDL/SpinalHDL #224: Multithreaded simulation with Verilator 4

rpls, 2019-07-02:
I can't find any info on whether this has been tried. Verilator 4 now supports multithreaded simulation [(slides about it)](https://www.veripool.org/papers/Verilator_v4_Multithreaded_OrConf2018.pdf), which promises some speedup. From reading the manual, one needs to pass `--threads N` to verilator to enable this. This will also cause thread-safe code to be emitted (for the most part), so there is a performance penalty when using only one thread. Also some features, like tracing, require a bit more care. Furthermore, I'd assume that there might be some problems with the JVM, which may not work together with such threads without some care.

Dolu1990, 2019-07-02:
Hi,

I didn't tried it, as currently i do not have designs big enough to realy take advantage of multi threading (from my estimation).

I have to check the verilator API changes. Do it change when this option is used ? As far my understanding goes, it still provide a eval function. So the JVM binding should work out of the box.

On the SimConfig. you can normaly call addSimulatorFlag to add verilator flags.
Maybe it should already be enough to get it work, not sure.

rpls, 2019-07-02:
> On the SimConfig. you can normaly call addSimulatorFlag to add verilator flags. Maybe it should already be enough to get it work, not sure.

Didn't see that, thanks for the tip. I just tried this with a simulated VexRiscv core on my laptop. Nothing exploded and the simulation seems to work correctly. However, the simulation took more than 3x as long...

I'm going to test this again on a proper machine and close the issue, if that doesn't show any speedup.

Dolu1990, 2019-07-02:
Right, i think you realy need a large and slow simulation to get speed up, Also i had the same kind of issue with the new verilator wave compression format. Reaaaaaly slow for small to medium simulations Much slower than VCD

Dolu1990, 2019-07-02:
But just adding that option as i sugested worked ??

rpls, 2019-07-02:
Yes, just adding the option seemed to work. At least, more threads are spawned and the aforementioned slowdown happened, so I assume it "worked". The simulated core also worked, so no errors as far as I can tell.

I also just tried it on a bigger machine with 18 threads, slowdown was MUCH worse... seems like the more threads, the worse it is. This makes me wonder, whether this might be a sideeffect of the JNI/JVM construction/interaction. But this is just a wild guess, I'm *very* far from an expert on JVM, Verilator & co. The only thing I might try in future is a pure Verilator / C++ testbench for another Verilog project of similar size to a RiscV core, just to see if this a problem concerning SpinalHDL/JVM/etc... But this is outside of the scope of this issue.

As far as I'm concerned, you can close the issue, unless you want others to have a look.

Dolu1990, 2019-07-02:
Let's me know how things goes with raw verilator :D
Also, maybe we can try that on the VexRiscv regression, as it is raw Verilator sim.
We can close the issue
Thanks anyway ^^

closed 2019-07-02 by Dolu1990