SpinalHDL/SpinalHDL #225: wrong documentation for Apb3UartCtrl

FrankBuss, 2019-08-10:
I think the [register mapping table](https://spinalhdl.github.io/SpinalDoc/spinal/examples/memory_mapped_uart/) is wrong. At least when I use the instructions [here](https://github.com/SpinalHDL/VexRiscv#murax-soc), I can use the sample code from VexRiscvSocSoftware, see [this file](https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/master/libs/uart.h) and for example the address 0 is the data word. When it is written, then data is sent with the UART, which is working.

Dolu1990, 2019-08-10:
Hi,

This register mapping table come from an example which isn't related the the uart controller in the Spinal.lib

FrankBuss, 2019-08-10:
Ok, this is confusing, especially because the documentation example has the exact same name as used in Murax.scala. Is the Apb3UartCtrl class which is provided by the SpinalHDL library, somewhere documented? I looked in the source code and it was not obvious to me how the address mapping is. The example code in the documentation uses different code, where the addresses are explicitly specified.

Dolu1990, 2019-08-12:
hmmm, it isn't well documented sorry for that, but there is the drivers : 
https://github.com/SpinalHDL/SaxonSoc/blob/Bmb/software/standalone/driver/uart.h

Else the source code which map the memory bus. For example the interrupts of the controller : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/com/uart/UartCtrl.scala#L126

For instance : 
 val readIntEnable  = busCtrlWrapped.createReadAndWrite(Bool, address = 4, 1) init(False)
Map the read interrupt enabled at address 4 bit 1   as a read/write register

FrankBuss, 2019-08-12:
Thanks, meanwhile I got it working by looking at the demo here:
https://github.com/SpinalHDL/VexRiscvSocSoftware/blob/master/libs/uart.h
Works in Verilator, and in Zephyr with a driver I wrote for it, only polling at the moment.

closed 2020-02-05 by Dolu1990