SpinalHDL/SpinalHDL #226: ClockDomain Synchronized Inheritance Problem

jijingg, 2019-08-13:
At first , ClockDomain may be used too casually. Most of the time we deal with Clock, not Domain. Considering the compatibility of SpinalHDL , let's use ClockDomain as Clock, the physical domain is replaced by its root. therefore:

- When create a ClockDomain, must inherit Root from its parent node, otherwise Root is itself.
- If we think the two Clocks are synchronized, we'll see if they have the same Root.
[image]

the setSyncWith method should not be encouraged, it make the problem complexing , if we setSyncWidth , first We need creat a vitural Root for two of there family。then Traverse all created clock domains and modify their Root to this virtual ClokDomain. I think it' not necessary.
[image]

If two clockDomain are sync, in Physical the must have same source. 
so we encourage the disigneer creat a Root-source clockDomain themself . 
[image]

Frequency division is slightly more complicated. If two ClockDomains only inherit Root, different frequency division factors may lead to asynchrony. 
[image]

to judge whether two clock domains are synchronized or not, it is necessary to find that is  the two ClockDomains dividing-factor can be divide with no remainder each other.
```
def isSyncWith(that : ClockDomain): Boolean {
  (this.root == that.root) && dividWithNoRemainder(this.divFactor, that.divFactor)
}
```

so to achive that , Maybe we should add 2 attribule  to ClockDomain 
- root
- divFactor

And method 

- gatedWith/gatedWithBlackBox
- inheritFrom(ClockDomain)
- divideSync(2)/divideASync(6)

ClockDomain created by gatedWith or divideSync need inherit Root and divFactor(parent.divFactor * currentdivFactor)  from Parents.

There is an exception the method divideASync will creat new CrossDomain .

Dolu1990, 2019-08-13:
Thanks for the issue :)

So this can be splited in two different issues : 
- The low level, which would be having something more flexible than setSyncWith. 
- The high level, as gatedWidth divideSync, ...

**For  the low level**
maybe each ClockDomain can be part of multiple syncronus groups : 
```scala
class SyncronusGroup{}
class ClockDomain{
   ...
   val syncGroups = ArrayBuffer[SyncronusGroup]()
}
```

This would allow to have a full flexibility on when you want or not to inheritFrom(X). It would also fix the actual issue of setSyncWith which can't coop with groups unless you manualy specify each combination.

**For the high level**

gatedWith/gatedWithBlackBox
inheritFrom(ClockDomain)
divideSync(2)/divideASync(6)

=> I agree

Would that be fine ? (especialy the low level)

jijingg, 2019-08-13:
Yes，I agree with you to divide it into two.   SyncGroup is also a good idea , it's flexible and easy to use for users .

Dolu1990, 2019-08-13:
Then one question is : 
What about 
```scala
val clk = in Bool()

val cdA = ClockDomain(clk)
val cdB = ClockDomain(clk)
```

Actualy, there is a phase in SpinalHDL which walk the clock signals untils their initial source to figure out automaticaly if things with different ClockDomain definitions are syncronus or not.

So an alternative / complement to ClockingGroup, could be to add a special tag to clocking signal to specify syncronisity

```scala
val clkA = in Bool()
val gatedClkA = clkA && cond
gatedClkA.addTag(SyncronusTo(clkA))  //stupid syntax, but get the idea
```

Then, the cross clock checking algo will automaticaly analyse clock paths and figure things out.

jijingg, 2019-08-14:
It's ok,  more way provided for users.  But in general, Inheritance is more recommended.

Dolu1990, 2019-08-17:
I'm looking at implementing that. One question is 

[image]

If that a2 is syncronus with c2, should that merge the whole syncFamilyA with the syncFamilyC ?

jijingg, 2019-08-20:
@Dolu1990 I'm sorry for havn't noticted the message in time.
The answer is Yes, if a2 sync to c2, their family must have same source .
- one way: the users creat root themselves
- way two: the Spinal auto insert a same root  when setSync occured

Dolu1990, 2019-08-24:
wip in the clockSyncRework branch, it as all the basic primitives of the first phase of that issue, enough to solve the second part.  especialy https://github.com/SpinalHDL/SpinalHDL/commit/21db6113721b100f278af2e5d7d449c22a9be42c#diff-2e250a329a7a1bd3ae374b013a8153efR163

Dolu1990, 2019-08-24:
The rule is as following.
Given two register using clk A and clk B
1) The checker, will collect all the clock chain which drive both A and B (and you can use Clock.syncDrive to jump over clock gating)
2) For each chain, are added the clock signals specified with setSynchronousWith.
3) Then if the intersection of the chain A and the chain B isn't empty => it is syncronus

jijingg, 2019-08-25:
@Dolu1990  thank you, It's great, with those two basic primitives of `setSynchronousWith` `syncDrive` ,  most scenarios can be covered.  It also easy to create some high-level use base on those primitives, such as clock-gate etc ..

closed 2020-02-05 by Dolu1990