SpinalHDL/SpinalHDL #227: Statemachine lost state name

Snoopy87, 2019-08-21:
Hi,

I would like to group some states with an Area as follow : 

```scala
  val sm = new StateMachine{

    val sIdle: State = new State with EntryPoint {
      whenIsActive{
        when(io.start){
          goto(sCheck.sEntry)
        }
      }
    }

    val sCheck = new Area{
      val sEntry: State = new State{
        whenIsActive{
          when(io.cmd(31 downto 16) =/= 0){
            goto(s1)
          }otherwise{
            goto(sErrorCmd)
          }

        }
      }
      val s1: State = new State{
        whenIsActive{
          goto(sKeyOp)
        }
      }
      val sErrorCmd:State = new State{
        whenIsActive{
          goto(sIdle)
        }
      }
    }

    val sKeyOp: State = new State{
      whenIsActive{
        goto(sDataOp.sWrite1)
      }
    }

    val sDataOp = new Area{
      val sWrite1: State = new State{
        whenIsActive{
          goto(sIdle)
        }
      }
    }
  }
```

Unfortunately, when I use Area, I lost the name of the State in my RTL which is bad for debugging

```vhdl
  function pkg_tosm_enumDefinition_defaultEncoding (value : std_logic_vector(2 downto 0)) return sm_enumDefinition is
  begin
    case value is
      when "000" => return boot;
      when "001" => return sm_sIdle;
      when "010" => return e2;
      when "011" => return e3;
      when "100" => return e4;
      when "101" => return sm_sKeyOp;
      when "110" => return e6;
      when others => return boot;
    end case;
  end;
````

numero-744, 2022-08-21:
I generated VHDL from your code and got this:

```vhdl
  function pkg_tosm_enumDef_native (value : std_logic_vector(2 downto 0)) return sm_enumDef is
  begin
    case value is
      when "000" => return BOOT;
      when "001" => return sIdle;
      when "010" => return sEntry;
      when "011" => return s1;
      when "100" => return sErrorCmd;
      when "101" => return sKeyOp;
      when "110" => return sWrite1;
      when others => return BOOT;
    end case;
  end;
```

So it seems to have been fixed, I think the issue can be closed.

Dolu1990, 2022-08-22:
hmm i guess the issue is more about how the sm area get its name ?
a easy way to check that is to do
this.setName("miaou") to see if things get named.

In which context this val sm is stored ?

Dolu1990, 2022-08-22:
Ahhh damned, i completly losed track of that issue XD
my bad.

numero-744, 2022-11-16:
@Dolu1990 Maybe it is because of `@dontName`s in `StateMachine.scala`?

Dolu1990, 2022-11-16:
>  Maybe it is because of @dontNames in StateMachine.scala?

I don't think so, probably the issue was due to something else.
Can't reproduce.

numero-744, 2022-11-16:
```scala

object TestFsm extends App {
  class Dut extends Component {
    val io = new Bundle {
      val start = in Bool ()
      val cmd = in Bits (32 bits)
    }

    import spinal.lib.fsm._
    val sm = new StateMachine {
      val sIdle: State = new State with EntryPoint {
        whenIsActive {
          when(io.start) { goto(sCheck.sEntry) }
        }
      }

      val sCheck = new Area {
        val sEntry: State = new State {
          whenIsActive {
            when(io.cmd(31 downto 16) =/= 0) { goto(s1) }
              .otherwise { goto(sErrorCmd) }
          }
        }
        println(sEntry.isNamed)
        println(sEntry.getPartialName())

        val s1: State = new State {
          whenIsActive { goto(sKeyOp) }
        }
        val sErrorCmd: State = new State {
          whenIsActive { goto(sIdle) }
        }
      }

      val sKeyOp: State = new State {
        whenIsActive { goto(sDataOp.sWrite1) }
      }

      val sDataOp = new Area {
        val sWrite1: State = new State {
          whenIsActive { goto(sIdle) }
        }
      }
    }
  }

  SpinalVhdl(new Dut)
}
```

There is no `sCheck` in the output.

numero-744, 2022-11-21:
Another option could be to document that behavior, that state names do not take into account areas XD

Dolu1990, 2022-11-23:
All good now ^^

commit b6c8f360 (2022-11-23): fix #227 Statemachine states are now named relatively to the StateMachne name
closed 2022-11-23 by Dolu1990