SpinalHDL/SpinalHDL #230: Weird warning "LATCH DETECTED" when building a priority encoder using muxList

gyn, 2019-09-02:
@Dolu1990 I want to use muxList to build a priority encoder based on SpinalTemplateSbt. When the width is 11 and less, it could build and produce right verilog file, but when the width becomes 12 and above, a warning "LATCH DETECTED" occurs. 
I tried to figure out why it happens, but failed. Please help check my code and/or others.
[SpinalTemplateSbt.tar.gz](https://github.com/SpinalHDL/SpinalHDL/files/3565003/SpinalTemplateSbt.tar.gz)
Thanks.
Yanan

Dolu1990, 2019-09-03:
Hi,
That's because with more than 4095 elements into a switch statement, you hit a limit into the 
https://github.com/SpinalHDL/SpinalHDL/blob/52be3be4ba2f243873d37b26b03fc998b3b3edf4/core/src/main/scala/spinal/core/internals/Statement.scala#L494  implement which is used to check if a switch statement cover all cases and transform the last "is"into a "default" one (and avoid latches) as implemented there : 
https://github.com/SpinalHDL/SpinalHDL/blob/7667363f54f0e6494c8faf06658278feecd9c987/core/src/main/scala/spinal/core/internals/Phase.scala#L1456

Currently as a work around, just do 
```scala
  io.encode := Vec(for (index <- 0 until limit) yield U(ffs(index))).read(U(io.request)).asBits
```

This avoid infering the mux via a SpinalHDL switch statement and directly use a SpinalHDL mux.

gyn, 2019-09-05:
@Dolu1990 Thanks. It works for me. Maybe I need to learn some scala programming first because I still confused about how this workaround works.

gyn, 2019-09-06:
@Dolu1990 So could I suggest add some descriptions about this limitation in SpinalHDL documents if you think it is a real limitation?

Dolu1990, 2019-09-07:
There is is https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Design%20errors/combinatorial_loop.html#false-positive
 :)

gyn, 2019-09-09:
@Dolu1990 Sorry. :). I am talking about the limitation of no more than or equal 4096 elements for a muxList.

closed 2022-10-19 by numero-744