SpinalHDL/SpinalHDL #231: Unexpected "COMBINATORIAL LOOP" error reported

gyn, 2019-09-05:
@Dolu1990 I am trying to using Array to build a 4-BCD counter. The project is 
[SpinalSandbox.tar.gz](https://github.com/SpinalHDL/SpinalHDL/files/3577308/SpinalSandbox.tar.gz)
I run sbt "runMain misc.array.BCDCellMain" and then check the produced verilog file, everything seems OK. When I run sbt "runMain misc.array.BCDMain", "COMBINATORIAL LOOP" error is reported. At first, I thought this may be caused by the Array thing, I just expanded the loop, created two BCDCell but later I saw same error.
Please correct me if I wrote code wrong, or give suggestions.
Thanks.
Yanan

Dolu1990, 2019-09-05:
Basicaly, it is because your code is using a Bits to get carry values in and out. Which result into some bits of a signal depending of some other bits of the same signals.
It isn't technicaly speaking a combinatorial loop, but is an issue for some simulators.

=> 
  val coBits = Vec(Bool, width)

also

you probably wanted 
BCDArray(0).io.ci  := False

Isn't it ?

Dolu1990, 2019-09-05:
(   Vec(Bool, width) will creat one Bool signal width times  )

gyn, 2019-09-06:
Hi Dolu,
Thanks for your suggestions. And also I fixed other issue, the final result is like:
[SpinalSandbox.tar.gz](https://github.com/SpinalHDL/SpinalHDL/files/3582692/SpinalSandbox.tar.gz)
I am learning verilog and also SpinalHDL by reading a book and trying finishing the exercises. And one of the exercises, I should use BCDArray(0).io.ci := True.
Yanan

closed 2019-09-06 by gyn