SpinalHDL/SpinalHDL #234: Hardware model not updated within current cycle when input changes during simulation

JobNoorman, 2019-10-03:
Reading about how the simulation engine [works](https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Simulation/engine.html), I expected that the Verilator model is updated *immediately* (i.e., within the current cycle) whenever a signal is changed from Scala.

However, this doesn't seem to be the case when running the following example:

```scala
class Test extends Component {
  val i = in Bool
  val o = out Bool
  val counter = Reg(UInt(5 bits)).init(0)
  counter := counter + 1
  o := counter === 5
}

object Test {
  def main(args: Array[String]) {
    SimConfig.withWave.compile(new Test).doSim { dut =>
      dut.clockDomain.forkStimulus(10)

      for (i <- 0 to 100) {
        dut.clockDomain.waitSampling()
        dut.i #= dut.o.toBoolean
      }
    }
  }
}
```

I would expect that the `dut.i` signal is always equal to `dut.o`. However, whenever `dut.o` changes, this change is only reflected in `dut.i` after one cycle.

Is this a bug or am I doing something wrong?

Dolu1990, 2019-10-03:
Basicaly, SpinalSim emulate on the top of verilator the same behaviour than what you have with a standard simulator.
The code executed after the dut.clockDomain.waitSampling() bloc will be : 
- After the clock edge
- But before the flipflop had time to put their input on their output

So dut.i will get the old value of the dut.o

I do not recommand to do the following code, but it would behave more or less as you wanted : 
        dut.clockDomain.waitSampling()
        sleep(1) //Give enough time for the FF to propagate
        dut.i #= dut.o.toBoolean

JobNoorman, 2019-10-03:
Are you saying that there's no proper way to simulate asynchronous logic in SpinalHDL? When using Verilator, I solve this issue by calling `eval()` every time C++ code changes any signals and the [flow diagram](https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Simulation/engine.html) seemed to imply that SpinalHDL does the same thing. If not currently possible, would you agree this is desirable to support? If so, I'd be willing to investigate how to implement this.

Dolu1990, 2019-10-03:
doing the way you said in verilator would not work for cases where the dut has outputs combinatorialy driven from inputs. I mean, you would have to eval again eval to be sure of everything.
What i mean, is that with raw verilator, there is no proper/clean way to manage combnatorial interractions between the testbench and the dut.

In spinalsim you can do such things.

So basicaly, you have to use forkSensitive : 

```scala
object PlayDevSpinalSim2 extends App{
  import spinal.core.sim._
  //Extend the functionalty of the SpinalSim forkSensitive to  only call the block of code if the trigger value changed. 
  //This avoid having endless simulation loop (dut write signal =>  rtl eval => dut write signal => ....  with no time progression)
  def forkSensitiveTrigger(trigger : BaseType)(block : => Unit): Unit = {
    var valueLast = trigger.toBigInt
    forkSensitive{
      val valueNew = trigger.toBigInt
      if(valueNew != valueLast) { block }
      valueLast = valueNew
    }
  }

  SimConfig.withWave.compile(new Component {
    val a = in UInt(8 bits)
    val b = out UInt(8 bits)
    val c = in UInt(8 bits)
    val d = out UInt(8 bits)
    b := RegNext(a) init(0)
    d := c
  }).doSim { dut =>
    dut.clockDomain.forkStimulus(period = 10)

    forkSensitiveTrigger(dut.b){
      dut.c #= dut.b.toInt
    }
    forkSensitiveTrigger(dut.d){
      dut.a #= (dut.d.toInt + 1) & 0xFF
    }

    dut.clockDomain.waitSampling(40)
  }
}
``` 

[image]

functionality as forkSensitiveTrigger should be added into the SpinalSim base API.

JobNoorman, 2019-10-04:
Oh, that's nice! And thanks for the quick commit! I never really understood the `forkSensitive` API so I didn't know you could use it for this.

Would it make sense to create a nice wrapper around `forkSensitiveTrigger`? Something like this:

```scala
implicit class ForkSensitiveTriggerPimper(lhs: BaseType) {
  def @=(rhs: BaseType): Unit = {
    forkSensitive(rhs) {
      lhs.assignBigInt(rhs.toBigInt)
    }
  }
}
```

That way, my original example can be written like this:

```scala
def main(args: Array[String]) {
  SimConfig.withWave.compile(new Test).doSim { dut =>
    dut.clockDomain.forkStimulus(10)

    for (i <- 0 to 100) {
      dut.clockDomain.waitSampling()
      dut.i @= dut.o
    }
  }
}
```

So the semantics of `@=` would be that the value will be assigned immediately while `#=` assigns at the next active clock edge. I guess we should also add overloads for `@=` to assign Scala types.

JobNoorman, 2019-10-04:
I tried implementing `@=` for the Scala Boolean type but ran into problems.

This was my first attempt:

```scala
implicit class ForkSensitiveTriggerBoolPimper(lhs: Bool) {
  def @=(rhs: Boolean): Unit = {
    forkSensitive(rhs) {
      lhs #= rhs
    }
  }
}
```

In this case, the assignment never gets executed because the if-test you [added](https://github.com/SpinalHDL/SpinalHDL/commit/36af4fc2156f52c62fd56e13bb0eb526fbe28bde#diff-b3e8acee07a9e906ecf46d5020a93642R163) is never true (`valueNew` always equals `valueLast`).

My second attempt was the following:

```scala
implicit class ForkSensitiveTriggerBoolPimper(lhs: Bool) {
  def @=(rhs: Boolean): Unit = {
    forkSensitiveWhile {
      lhs #= rhs
      false
    }
  }
}
```

But in this case, the assignment is again only visible at the next cycle (and I don't understand why).

Any idea why this doesn't work?

JobNoorman, 2019-10-04:
I think I misunderstood what `forkSensitive` actually does and came to the conclusion that my suggestion about adding `@=` probably doesn't make sense :smile:

Am I correct in saying the `forkSensitive(trigger)` is the equivalent of doing `always @(trigger)` in Verilog?

Dolu1990, 2019-10-21:
All good now right ?

closed 2019-10-21 by Dolu1990

JobNoorman, 2019-10-21:
> Yes right ^^
> All good now right ?

Yes, all good :smile: Thanks!