SpinalHDL/SpinalHDL #236: T(0 bits) Problem

jijingg, 2019-10-18:
hi @Dolu1990 , i'm doing some lib and need use `(0 bits)`, found some problem, seems like bugs.
## CASE 1
```scala
class T1 extends Component{
    val a       = Reg(Bits(2 bits)) init 0 
    val nonBit  = Bits(0 bits)   

    val o1 = nonBit ## a ## nonBit    // generator error verilog
}
```
generate Error Verilog
```verilog
module T1 (
      input   clk,
      input   reset);
  wire [1:0] a;
  wire [1:0] o1;
  assign a = (2'b00);
  assign o1 = {nonBit  ,a};  //!! here nonBit is not needed, and cause Error!
endmodule
```
## CASE 2
```scala
class T2 extends Component{
    val a       = Reg(Bits(2 bits)) init 0 
    val nonBit = Bits(0 bits).setAll()

    val o2 = a ## Bits(0 bit)                //pass
    val o3 = Bits(0 bit) ## a                //pass                        
    val o5 = Bits(0 bit) ## a ## Bits(0 bit) //cause exception
}
```
`val o1 = nonBit ## a ## nonBit` generate pass
`val o4 = Bits(0 bit) ## a ## Bits(0 bit)` cause exception

Logs:
```shell
**********************************************************************************************
[Warning] Elaboration failed (0 error).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 3054.681 : Elaborate components
[Progress] at 3054.681 : Checks and transforms
[Progress] at 3054.683 : Generate Verilog
java.lang.Exception: Internal error
  spinal.core.Nameable.getNameElseThrow(Trait.scala:409)
  spinal.core.Nameable.getNameElseThrow$(Trait.scala:407)
  spinal.core.BaseType.getNameElseThrow(BaseType.scala:52)
  spinal.core.internals.ComponentEmitterVerilog.$anonfun$emitReference$1(ComponentEmitterVerilog.scala:697)
  scala.collection.MapLike.getOrElse(MapLike.scala:131)
  scala.collection.MapLike.getOrElse$(MapLike.scala:129)
  scala.collection.AbstractMap.getOrElse(Map.scala:63)
  spinal.core.internals.ComponentEmitterVerilog.emitReference(ComponentEmitterVerilog.scala:697)
  spinal.core.internals.ComponentEmitterVerilog.refImpl(ComponentEmitterVerilog.scala:1116)
  spinal.core.internals.ComponentEmitterVerilog.dispatchExpression(ComponentEmitterVerilog.scala:1249)
  spinal.core.internals.ComponentEmitterVerilog.emitExpression(ComponentEmitterVerilog.scala:722)
  spinal.core.internals.ComponentEmitterVerilog.operatorImplAsCat(ComponentEmitterVerilog.scala:1145)
  spinal.core.internals.ComponentEmitterVerilog.dispatchExpression(ComponentEmitterVerilog.scala:1310)
  spinal.core.internals.ComponentEmitterVerilog.emitExpression(ComponentEmitterVerilog.scala:722)
  spinal.core.internals.ComponentEmitterVerilog.emitAsynchronous(ComponentEmitterVerilog.scala:407)
  spinal.core.internals.ComponentEmitterVerilog.$anonfun$emitArchitecture$16(ComponentEmitterVerilog.scala:179)
  spinal.core.internals.ComponentEmitterVerilog.$anonfun$emitArchitecture$16$adapted(ComponentEmitterVerilog.scala:175)
  scala.collection.mutable.LinkedHashSet.foreach(LinkedHashSet.scala:95)
  spinal.core.internals.ComponentEmitterVerilog.emitArchitecture(ComponentEmitterVerilog.scala:175)
  spinal.core.internals.ComponentEmitterVerilog.<init>(ComponentEmitterVerilog.scala:1377)
  spinal.core.internals.PhaseVerilog.compile(PhaseVerilog.scala:72)
  spinal.core.internals.PhaseVerilog.$anonfun$impl$1(PhaseVerilog.scala:49)
  spinal.core.internals.PhaseVerilog.$anonfun$impl$1$adapted(PhaseVerilog.scala:46)
  scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
  scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
```

Dolu1990, 2019-10-21:
I will check that, thanks for the report :)

Dolu1990, 2019-10-22:
Got it, basicaly, i have a function that i used to remap expression giving a lamda function.

the issue, is that if a expression is remapped, it do not redo the remaping to see if there again a new symplification possible.

when i mean remaping, it is for instance remplacing the ## by its left or right operand.
I'm trying to find a nice way to fix that

Don't want to hack things everywere like that : 
```scala
  override def remapExpressions(func: (Expression) => Expression): Unit = {
    while(true) {
      val oldLeft = left
      val oldRight = right
      left = func(left).asInstanceOf[T]
      right = func(right).asInstanceOf[T]
      if(oldLeft == left && oldRight == right) return
    }
  }
```

instead of the original 
```scala
  override def remapExpressions(func: (Expression) => Expression): Unit = {
    left = func(left).asInstanceOf[T]
    right = func(right).asInstanceOf[T]
  }
```

closed 2019-10-22 by Dolu1990

Dolu1990, 2019-10-22:
Nice catch :D

commit 58ecf71d (2019-10-22): Fix #236