SpinalHDL/SpinalHDL #237: A new RegisterBank Solution:  RegIf

jijingg, 2019-10-28:
hi @Dolu1990, I tried to redesign the register interface scheme
There are three major updates
- Automatic address and fields allocate, and conflict detection
- 28 Register Access types(Covering 25 types of UVM)
- Auto Document generate

## Automatic allocate 

Automatic address allocate 
```scala
  class RegBankExample extends Component{
    val io = new Bundle{
      apb = Apb3(Apb3Config(16,32))
    }
    val busSlave = BusInterface(io.apb,(0x0000, 100 Byte)
    val M_REG0  = busSlave.newReg(doc="REG0")
    val M_REG1  = busSlave.newReg(doc="REG1")
    val M_REG2  = busSlave.newReg(doc="REG2")

    val M_REGn  = busSlave.newRegAt(address=0x40, doc="REGn")
    val M_REGn1 = busSlave.newReg(doc="REGn1")
  }
```
[image]

Automatic fileds allocate 
```scala
  val M_REG0  = busSlave.newReg(doc="REG1")
  val fd0 = M_REG0.field(2 bits, RW, doc= "fields 0")
  M_REG0.reserved(5 bits)
  val fd1 = M_REG0.field(3 bits, RW, doc= "fields 0")
  val fd2 = M_REG0.field(3 bits, RW, doc= "fields 0")
  //auto reserved 2 bits
  val fd3 = M_REG0.fieldAt(pos=16, 4 bits, doc= "fields 3")
  //auto reserved 12 bits
```
[image]

confilict detection 
```scala
  val M_REG1  = busSlave.newReg(doc="REG1")
  val r1fd0 = M_REG1.field(16 bits, RW, doc="fields 1")
  val r1fd2 = M_REG1.field(18 bits, RW, doc="fields 1")
...
  cause Exception
```
```scala
  val M_REG1  = busSlave.newReg(doc="REG1")
  val r1fd0 = M_REG1.field(16 bits, RW, doc="fields 1")
  val r1fd2 = M_REG1.field(offset=10, 2 bits, RW, doc="fields 1")
...
  cause Exception
```
##  28 Access Type 

Most of these come from UVM

| AccessType | Description                                                                     | Status |
|------------|---------------------------------------------------------------------------------|--------|
| RO         | w: no effect, r: no effect                                                      | UVM    |
| RW         | w: as-is, r: no effect                                                          | UVM    |
| RC         | w: no effect, r: clears all bits                                                | UVM    |
| RS         | w: no effect, r: sets all bits                                                  | UVM    |
| WRC        | w: as-is, r: clears all bits                                                    | UVM    |
| WRS        | w: as-is, r: sets all bits                                                      | UVM    |
| WC         | w: clears all bits, r: no effect                                                | UVM    |
| WS         | w: sets all bits, r: no effect                                                  | UVM    |
| WSRC       | w: sets all bits, r: clears all bits                                            | UVM    |
| WCRS       | w: clears all bits, r: sets all bits                                            | UVM    |
| W1C        | w: 1/0 clears/no effect on matching bit, r: no effect                           | UVM    |
| W1S        | w: 1/0 sets/no effect on matching bit, r: no effect                             | UVM    |
| W1T        | w: 1/0 toggles/no effect on matching bit, r: no effect                          | UVM    |
| W0C        | w: 1/0 no effect on/clears matching bit, r: no effect                           | UVM    |
| W0S        | w: 1/0 no effect on/sets matching bit, r: no effect                             | UVM    |
| W0T        | w: 1/0 no effect on/toggles matching bit, r: no effect                          | UVM    |
| W1SRC      | w: 1/0 sets/no effect on matching bit, r: clears all bits                       | UVM    |
| W1CRS      | w: 1/0 clears/no effect on matching bit, r: sets all bits                       | UVM    |
| W0SRC      | w: 1/0 no effect on/sets matching bit, r: clears all bits                       | UVM    |
| W0CRS      | w: 1/0 no effect on/clears matching bit, r: sets all bits                       | UVM    |
| WO         | w: as-is, r: error                                                              | UVM    |
| WOC        | w: clears all bits, r: error                                                    | UVM    |
| WOS        | w: sets all bits, r: error                                                      | UVM    |
| W1         | w: first one after ~hard~ reset is as-is, other w have no effects, r: no effect | UVM    |
| WO1        | w: first one after ~hard~ reset is as-is, other w have no effects, r: error     | UVM    |
| NA         | w: reserved, r: reserved                                                        | New    |
| W1P        | w: 1/0 pulse/no effect on matching bit, r: no effect                            | New    |
| W0P        | w: 0/1 pulse/no effect on matching bit, r: no effect                            | New    |

## Auto generate document 
Document Type 
- JSON
- reStructuredText 
- markdown 
- HTML
- docx 

HTML auto-doc is now complete, Example source Code:

https://github.com/jijingg/SpinalHDL/blob/regif/tester/src/main/scala/spinal/tester/code/RegIfExample.scala

generated HTML document:

[image]

bellaz89, 2020-05-09:
@jijingg 

You did a quite impressive work!
I have two questions regarding your interface:

1. Are ranged addresses supported? For example if I want to map a RAM or a ROM on the bus how can achieve this? Is there something similar to SizeMapping in SlaveFactory?
2. Is it possible to block the bus transaction to wait until some event/process is completed/ready?

Looking forward to your work!

jijingg, 2020-05-10:
> Are ranged addresses supported? For example if I want to map a RAM or a ROM on the bus how can achieve this? Is there something similar to SizeMapping in SlaveFactory?

Not yet，but i agree , a ranged address for RAM are quite  important requirement. it easy to wire out a port for RAM,ROM, but the document for RAM address field may be more complex，

1. the two address of a ram maybe have difference field mapping   
2. the address mapping may repeat by group of different Number

so the current register interface document format not suit for this situation .   

I am thinking about making a more general address mapping Document format. With the help of HTML hyperlink, we can view the specific content of the address space through  click.
[image]

But the detail is still under consideration.

jijingg, 2020-05-10:
> Is it possible to block the bus transaction to wait until some event/process is completed/ready?

There should be no problem in principle, but could you give me a real example for detail .
@bellaz89

Dolu1990, 2020-05-10:
@jijingg Hoo one thing which is quite new now and which can be usefull is that now, SpinalHDL can give proper names to things right out of the box, without tricks 

```scala
val x = new Thing extends Nameable
println(x.getName())
```

jijingg, 2020-05-10:
@Dolu1990 good, i will it later

bellaz89, 2020-05-11:
@jijingg 

> > Is it possible to block the bus transaction to wait until some event/process is completed/ready?
> 
> There should be no problem in principle, but could you give me a real example for detail .
> @bellaz89

For example, for the interface I am doing right now for a coprocessor attached to an Apb3 bus through the BusSlaveFactory I have the following use cases:

**Writing to a Stream-like mapped component that may not be ready to accept data:**

[image]

In this case "value" has to be written to the component. However, the bus interface has to wait for the "ready" line to be asserted. Therefore the transaction scheme is the following:

1. set "value" with the value provided from the bus. Set "valid" to True for the transaction duration
2. block until the "ready" line is asserted.
3. finish the transaction

**Reading from a mapped component that takes multiple cycles to generate data or may not be ready:**

[image]

In this case "value" has to be generated and read from the component. To do this, a request has to be issued. Then the bus interface has to wait until the generated value is valid. The component may be not ready to accept requests.  Therefore the transaction scheme is the following:

1. set "ask_value" to True
2. block until the "ready" line is asserted.
3. set "ask_value" to False
4. block until "value_valid" is True
5. put "value" to the bus and finish the transaction

I implemented these two kinds of transactions using primitives of BusSlaveFactory (like [read/write]Halt, or on[Read/Write]Primitive).

Thanks!

jijingg, 2020-05-12:
what you talk about do we need handshake on RegisterBank , right ?
I haven't do handshake in the BusIf and assume that the register is always valid when bus read and write not matter what value it is.  The correctness of register value is guaranteed in high level like interrupt with proper hw design . 
for example .  at every slot we need read last slot calc result ,  if we halt and wait the register valid , that will too time expensive . in the halt time nothing can do!
[image]
so the right way HW lock at right time , and SW read at every interrupt.
[image]
so  i think  the  handshake of ready/valid  is something for HW to HW ,
in the case of SW to HW ,we need to design and prepare proper data when SW read occur, not halt or blocking . right? @bellaz89

bellaz89, 2020-05-12:
@jijingg 

Hi!

For sure if the bus is blocked no more work can be done on the CPU side and this can be wasteful if several clock cycles are required for the operation to complete. However, I don't know if timewise using only interrupts improves the speed since they also need time to be processed + the time to access the bus. In the coprocessor I am developing, the memory-mapped read/writes block the bus at maximum for 4-5 clock cycles. I am unsure if using interrupts is going to consume fewer clock cycles than this. Maybe @Dolu1990  can give more hints about how much time is required to properly manage external interrupts + context switches in 'common' VexRiscV configurations.  

Thanks!

jijingg, 2020-05-12:
@bellaz89 

the RegIf mainly faces register or simple ram interface ,  My experience is mainly about IP design, If there is really complex MEM mapping, I will connect AHB to each MEM, at most readSync with delay 1 cycle.  so there have no problem.

but I don't know that in other places such as Process design (the frequency is high, so need divide many pipeline for good timing, in those case Delay maybe more than many cycle),  in this situation, we need handshake. 

in addition，the interrupt things what i'm talking is about SW interaction in IP，not for mapping delays  , so forget that XD

numero-744, 2022-07-25:
I think this issue can be closed as done

Dolu1990, 2022-08-02:
Hoo right, it got merged

closed 2022-08-02 by Dolu1990