SpinalHDL/SpinalHDL #239: Bits().asBool Problem

jijingg, 2019-10-31:
i found some problem on Bits().asBool
## CASE1  : error
```scala
class T1  extends Component{  
  val sel = in Bool()
  val a = Reg(Bits(1 bits)) init 0 
  when(sel){a.asBool.set()} 
}
showRtl(new T1)
```
got Error Verilog 
```verilog 
module T1 (
      input   sel,
      input   clk,
      input   reset);
  wire [0:0] a;
  assign a = (1'b0);
endmodule
```
## CASE2  : ok
`a(0)`  `a.lsb` all generate ok 
```scala
class T2  extends Component{  
  val sel = in Bool()
  val a = Reg(Bits(1 bits)) init 0 
  when(sel){a(0).set()} 
  // when(sel){a.lsb.set()}  //also Ok
}
showRtl(new T2)
```
got right Verilog 
```verilog 
module T2 (
      input   sel,
      input   clk,
      input   reset);
  reg [0:0] a;
  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      a <= (1'b0);
    end else begin
      if(sel)begin
        a[0] <= 1'b1;
      end
    end
  end
endmodule
```

Dolu1990, 2019-10-31:
Hi,
So it work as i expected. Basicaly, the lsb function is a redirection to the LSB bit, as the range access 
 => myBits(x downto y).
While the asBool return a copy of the LSB bits value, as all other cast operation.

It does not sound fair ?

jijingg, 2019-10-31:
ok, I know what you mean , the copy is another instance , can't treat as a part of origin Bits. 
So if i do something on origin Bits, the `a(0)`, `a.lsb` is recommended.    

   Some times `Bits(1 bits)(0)`,`Bits(1 bits).lsb` a little awkward，But it 's acceptable.   
   thank you

closed 2019-10-31 by jijingg