SpinalHDL/SpinalHDL #240: SpinalSimConfig.addRTL() has no effect

kaofishy, 2019-11-03:
While creating a simulation, I tried adding an external RTL source for a blackbox component using `SpinalSimConfig.addRTL()` and it appears to have no effect, Verilator keeps complaining that it cannot find the RTL file no matter where I put the file in the project tree.

Sample code:
```scala
    SimConfig.addRtl("DSP48E1.v").compile(new DSPTest).doSim{dsp =>
      dsp.clockDomain.forkStimulus(10)
      for(idx <- 0 until 100) {
        dsp.io.dataIn #= Random.nextInt(255)
        dsp.clockDomain.waitSampling()
      }
    }
```
Sample output:
```
[Progress] Verilator compilation started
%Error: ../../tmp/job_1/DSPTest.v:104: Cannot find file containing module: 'DSP48E1'
  DSP48E1 #( 
  ^~~~~~~
%Error: ../../tmp/job_1/DSPTest.v:104: This may be because there's no search path specified with -I<dir>.
  DSP48E1 #( 
  ^~~~~~~
        ... Looked in:
             DSP48E1
             DSP48E1.v
             DSP48E1.sv
             verilator/DSP48E1
             verilator/DSP48E1.v
             verilator/DSP48E1.sv
%Error: ../../tmp/job_1/DSPTest.v:186: Cannot find file containing module: 'DSP48E1'
  DSP48E1 #( 
  ^~~~~~~
%Error: Exiting due to 3 error(s)
```

Dolu1990, 2019-11-07:
In which folder relatively to the project is the DSP48E1.v ?

kaofishy, 2019-11-15:
@Dolu1990 I'm on Windows 10, using IntelliJ IDEA. I am able to use the Verilator simulation with SpinalHDL on its own, just not with an external RTL file.

Dolu1990, 2019-11-16:
ahh thats likely the issue (W10)
SpinalHDL is mainly developped and tested on Linux.
One current workaround that you can do is specifying extra verilator :

SimConfig.addSimulatorFlag("-Iabsolute_path_to_rtl")....

That should normaly be fine then.

Dolu1990, 2019-12-10:
For as reference, there is an example working on linux, which integrate a sd card verilog emulator into a SpinalSim, it has both .v and .h.

https://github.com/SpinalHDL/SaxonSoc/blob/dev/hardware/scala/saxon/board/ulx3s/Ulx3sLinux.scala#L259

I will test it on windows

kaofishy, 2019-12-11:
@Dolu1990 thanks for all the help, btw the `-Iabsolutepath` workaround worked for me in the meantime.

closed 2022-10-23 by numero-744