SpinalHDL/SpinalHDL #243: Cannot infer dual write port Memas blockram in Quartus

kaofishy, 2019-12-04:
I'm trying to implement a Mem with two write ports in Quartus and no matter what I've tried it keeps synthesizing the generated Verilog as registers rather than blockram.

For example, the following code will not generate Verilog that Quartus recognizes as blockram
```scala
  val port1 = ram.writePort
  val port2 = ram.writePort

  port1.address := addrCounter
  port1.data := 0x33333333
  port1.valid := io.state

  port2.address := addrCounter.valueNext
  port2.data := 0x77777777
  port2.valid := io.state

  val rport = ram.readSyncPort
  rport.cmd.payload := port1.address
  rport.cmd.valid := !port1.valid
```
I've also tried various combinations of `Mem.write()` and `Mem.readWriteSync()` to no avail. Am I doing something wrong here?

kaofishy, 2019-12-04:
Update: tried to synthesize the same code in Vivado, and got this warning:
```
WARNING: [Synth 8-4767] Trying to implement RAM 'ram_reg' in registers. Block RAM or DRAM implementation is not possible; see log for reasons.
Reason is one or more of the following :
	1: RAM has multiple writes via different ports in same process. If RAM inferencing intended, write to one port per process. 
	2: Unable to determine number of words or word size in RAM. 
	3: No valid read/write found for RAM. 
RAM "ram_reg" dissolved into registers
```
Looks like SpinalHDL combines both write ports into the same `always@` block:
```Verilog
  always @ (posedge clk) begin
    if(io_state) begin
      ram[addrCounter_value] <= _zz_4_;
    end
    if(io_state) begin
      ram[addrCounter_valueNext] <= _zz_5_;
    end
  end
```
When I separate the two assignments into different synchronous processes I was able to successfully infer dual write ports in both Quartus and Vivado.

kaofishy, 2019-12-10:
After some more experiments I discovered that if I create a dummy ClockDomain for the second write port SpinalHDL generates the correct inferable code:
```scala
  ram.write(addrCounter, 0x77777777, io.state)
  led_reg := ram.readSync(addrCounter)
  led_reg2 := ram.readSync(addrCounter.valueNext)
  new ClockingArea(ClockDomain(clockDomain.clock)) {
    ram.write(addrCounter.valueNext, 0x33333333)
  }
```

This results in two separate process statements in Verilog:
```verilog
  always @ (posedge clk) begin
    if(_zz_6_) begin
      _zz_1_ <= ram[addrCounter_value];
    end
  end

  always @ (posedge clk) begin
    if(_zz_7_) begin
      _zz_2_ <= ram[addrCounter_valueNext];
    end
  end
```

@Dolu1990 is there a less hacky way to accomplish this? How did you do this for the read ports, because right now the read ports are correctly in two separate process blocks, and reading through `readSyncImpl` and `writeImpl` I can't identify the difference.

Dolu1990, 2019-12-10:
Sorry for the delay, i'm on too many things at once XD

Ahhh, infering ram is realy a paine in the ass ^^
Depending synthesis tool that's a differnt story, but i think you are right, memory write ports should never be in the same process, to avoid infering a write ordering.

Will fix that, then release 1.3.7

Dolu1990, 2019-12-10:
ahh about 

```
  val port1 = ram.writePort
  val port2 = ram.writePort

  port1.address := addrCounter
  port1.data := 0x33333333
  port1.valid := io.state

  port2.address := addrCounter.valueNext
  port2.data := 0x77777777
  port2.valid := io.state

  val rport = ram.readSyncPort
  rport.cmd.payload := port1.address
  rport.cmd.valid := True
```

You can't infer in FPGA memory blocks a ram with 2 write ports and a read port, with all different address. I mean, to me, it seem not possible, best you could technicaly without using advanced architectural tricks is 1 write port + N read ports or 2 ReadWrite ports

Right ?

Dolu1990, 2019-12-10:
Ok, so, i made a update, basicaly, now the idea is that all the syncronus memory reads will have a "dontCare" readDuringWrite policy. As that's the most common and portably way of doing things (as far i know).

So now, Each read and each write operations on the memories will have their dedicated process, so no ordering can be then inferred by the synthesis, which should help mapping things :)

Now i just have to test how well this is infered on synthesis tools XD

Dolu1990, 2019-12-10:
This is my testbench : 

```
  class TopLevel extends Component{
    val a = new Area {
      val ram = Mem(Bits(8 bits), 1024)

      val write = slave(ram.writePort)
      val read = slave(ram.readSyncPort)
    }

    val b = new Area {
      val ram = Mem(Bits(8 bits), 1024)

      val write = slave(ram.writePort)
      val read_1 = slave(ram.readSyncPort)
      val read_2 = slave(ram.readSyncPort)
      val read_3 = slave(ram.readSyncPort)
    }

    val c = new Area {
      val ram = Mem(Bits(8 bits), 1024)

      val write = slave(ram.writePort)
      val readAddress = in UInt(10 bits)
      val readData = out(ram.readAsync(readAddress))
    }

    val d = new Area {
      val ram = Mem(Bits(8 bits), 1024)

      val address = in UInt(10 bits)
      val writeData = in Bits(8 bits)
      val readData = out Bits(8 bits)
      val enable = in Bool()
      val write = in Bool()
      readData := ram.readWriteSync(address, writeData, enable, write)
    }

    val e = new Area {
      val ram = Mem(Bits(8 bits), 1024)

      val address_1 = in UInt(10 bits)
      val writeData_1 = in Bits(8 bits)
      val readData_1 = out Bits(8 bits)
      val enable_1 = in Bool()
      val write_1 = in Bool()
      readData_1 := ram.readWriteSync(address_1, writeData_1, enable_1, write_1)

      val address_2 = in UInt(10 bits)
      val writeData_2 = in Bits(8 bits)
      val readData_2 = out Bits(8 bits)
      val enable_2 = in Bool()
      val write_2 = in Bool()
      readData_2 := ram.readWriteSync(address_2, writeData_2, enable_2, write_2)
    }
  }
``` 

Vivado is happy with it, inferation is just fine.

Dolu1990, 2019-12-10:
Quartus do not like the "b" ram XD Else that's fine

Dolu1990, 2019-12-11:
Will also add 

```
    val f = new Area {
      val ram = Mem(Bits(8 bits), 1024)

      val address_1 = in UInt(10 bits)
      val write_1 = in Bool()
      val dataWrite_1 = in Bits(8 bits)
      val dataRead_1 = out Bits(8 bits)
      ram.write(address_1, dataWrite_1, write_1)
      dataRead_1 := ram.readSync(address_1)

      val address_2 = in UInt(10 bits)
      val write_2 = in Bool()
      val dataWrite_2 = in Bits(8 bits)
      val dataRead_2 = out Bits(8 bits)
      ram.write(address_2, dataWrite_2, write_2)
      dataRead_2 := ram.readSync(address_2)
    }
```

Currently Quartus with Cyclone V has quite a few issue infering the ram, i will investigate futher tomorrow

kaofishy, 2019-12-11:
Yes, it looks like for some reason Cyclone V synthesis is a lot more strict on read-during-write behaviour than for Cyclone IV/10LP/MAX 10. I keep getting `RAM logic "ram" is uninferred due to unsupported read-during-write behavior` for code that otherwise synthesizes on the LUT4-based families.

kaofishy, 2019-12-11:
Update: so I applied `addAttribute("ramstyle", "no_rw_check")` on the ram instance and now Quartus happily synthesizes the same code on Cyclone V.

Dolu1990, 2019-12-13:
So i made some tests with the following targets : 
- Arty 7 (OK)
- ECP5 diamond (OK)
- Ice40 icestorm (OK)
- Cyclone  V => true dual port ram realy require the no_rw_check attribute :/

So tree solutions :
- Letting the user add the no_rw_check when he know the target is Cyclone V and the ram is a true dual ported one
- Adding some target (CycloneV) specification in the SpinalConfig which would automaticaly add such attributes
- Always adding that attributes, but that could have come conflicts when you use Xilinx stuff XD, so not realy a good solution

What do you think ?

Dolu1990, 2019-12-13:
Hooo also, Cyclone V, infering distributed memory (MLAB) do not work :/ don't know why

kaofishy, 2019-12-16:
Honestly I don't know if it's worth it to include the workaround by default since every single other synthesis tool had no issue (I even tried the Anlogic toolchain and it worked fine). Also in the process of researching the bug I stumbled upon this hilarity: [Why does my RAM inference fail for Stratix 10 designs?](https://www.intel.com/content/www/us/en/programmable/support/support-resources/knowledge-base/tools/2016/why-does-my-ram-inference-fail-for-stratix-10-designs-.html)

Dolu1990, 2019-12-16:
XD

So, got a fix, basicaly, no tag will be added by default, but you can do the following : 

```scala
SpinalConfig(device = Device.ALTERA).generateVerilog(new TopLevel())
```

And then  all ram with only read ports which are readDurringWrite=don'tCare (the default) will have the tag added, so synthesis will be happy.

commit 9fc38a6a (2019-12-16): Now Both VHDL and Verilog backend emit syncronus reads as don't care readDurrinWrite using separated process for each memory ports.
Add Device.ALTERA, which automaticaly add no_rw_check on don'tcare memories
fix #243
closed 2019-12-16 by Dolu1990