SpinalHDL/SpinalHDL #245: Enhancement:  B(value: Array[Byte][, x bits])

bellaz89, 2019-12-11:
I don't know if I am missing something but I found it hard to convert an arbitrary long word to Bits.
For this, I propose to add the following method  
```
B(value: Array[Byte][, x bits]): Byte
```
that fills value with the data contained in Array[Byte]
if x < value.length*8 then value is truncated. If x > value.length*8 then value is padded with zeros. if x is not passed the final length is value.length*8

This would be very useful IMHO in the case of data read from a binary file or other binary resources that need to be converted to a literal (for example to initialize a Mem..).

bellaz89, 2019-12-12:
@Dolu1990

Dolu1990, 2019-12-12:
Hmmm what's about Scala BigInt ?
You can assign them to signal in sim and rtl.

bellaz89, 2019-12-12:
Yes.   This fits. Maybe then a doc reference to this has to be added (there is no reference to this in https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Data%20types/bits.html) 

Thanks

Dolu1990, 2019-12-12:
Hoo right, it only refer to Int

bellaz89, 2019-12-12:
Also, is  B(value: BigInt) going to work with negative BigInts? As far as I remember last time I checked B(value: Int) didn't work with negative Ints. This could be a limitation for bit streams that starts with '1'

Dolu1990, 2019-12-12:
That's done on purpose that negative value only work with SInt

bellaz89, 2019-12-12:
Ahh ok, now I got it.. the right way to do it should be the following:

```
 B(BigInt(java.math.BigInteger(1, byteArray))) 
```

closed 2020-01-07 by bellaz89