SpinalHDL/SpinalHDL #246: Template project with Gradle

piegamesde, 2019-12-15:
Using SpinalHDL, I've come to learn that SBT is a strictly inferior build tool compared to Gradle (change my mind). So I took the Spinal base project and migrated it to use Gradle instead:

https://github.com/piegamesde/SpinalTemplateGradle

Would you accept integrating this into the SpinalHDL project account? I'd also update the README here on how to use it.

Dolu1990, 2019-12-19:
Hi,

Sure, there is https://github.com/SpinalHDL/SpinalTemplateGradle :D thanks !

piegamesde, 2019-12-20:
You just forked mine? :D I rather suggest to transfer ownership, if this is possible.

Dolu1990, 2019-12-20:
Hooo, hmmm, i don't know if there is a way to transfer ownership in github. Personnaly, i'm fine with the fork, you did the job after all :)

piegamesde, 2019-12-20:
Did the trick with deleting mine. (<https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/what-happens-to-forks-when-a-repository-is-deleted-or-changes-visibility>)

closed 2019-12-20 by piegamesde

Dolu1990, 2019-12-20:
RIP