SpinalHDL/SpinalHDL #247: Asserts don't work

piegamesde, 2019-12-19:
I want to put assertions into my hardware that get checked during simulation. But currently, I can only get simulation-level\* assertions to work. If I put

    assert(counter.value == 2, "This is fine")

into my design, this will fail during *evaluation*, and not during *simulation*. If I alternatively put

    GenerationFlags.formal {
      assert(counter.value == 2, "This is fine")
    }

into my code, the simulation will run through without ever failing, which is obviously not what should be happening.

\* What works: If I put an assertion into the `dut` testing code that drives the hardware, it will work as expected. But that's not really what I want.

Dolu1990, 2019-12-19:
Ahhh funny one, basicaly, because you were using the == operator (scala reference check) instead of the === (SpinalHDL hardware equality), scala was identifying the assert function as the native scala one, not the SpinalHDL one.

``` 
assert(counter.value === 2, "This is fine")
```
Should work just fine 

``` 

  always @ (posedge clk) begin
    if(!(counter_value == (5'b00010))) begin
      $display("ERROR This is fine");
    end
  end

```

piegamesde, 2019-12-20:
I suggest creating different methods for "simulation asserts" and "hardware asserts" to avoid confusion. This would also allow for more features in "simulation" asserts. (We all know and love them from JUnit)

piegamesde, 2019-12-20:
Okay, I changed it from `==` to `===`. While the assert triggers, it only prints an ERROR without failing the test. Better than nothing, but still not ideal.

Dolu1990, 2019-12-20:
Hmm basicaly, you can give a severity level as a third arguement. the default level is ERROR, but normaly, a FAILURE should stop the simulation.

hmm
Maybe that right, default should be failure ?

commit 60c3eea0 (2020-02-03): Change assertion severity

If an assertion fails, the simulation should stop. Otherwise, the user won't have a trace to see how this happened. Closes #247
closed 2020-02-03 by Dolu1990