SpinalHDL/SpinalHDL #250: Usage of hardened IP of ice40

Riconec, 2020-01-03:
Hi, I am interested to use SpinalHDL but some things not clear for me:
How I am supposed to infer non-digital HW for a specific device, e.g ice40up5k (hardened spi, rgb current source driver, pll and integrated oscillators)?

Here is the link to "technology library" where all available primitives described with inferring examples:

http://www.latticesemi.com/~/media/LatticeSemi/Documents/TechnicalBriefs/SBTICETechnologyLibrary201504.pdf

One way that I can figure out is creating verilog files with specific primitives inferred and connect them with scala files on synthesis step, but I hope there is a way to add "inline verilog" to spinalHdl

Dolu1990, 2020-01-03:
Hi,

Basicaly, it is a bit like a C header file, you can define blackbox in SpinalHDL : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/lib/src/main/scala/spinal/lib/blackbox/lattice/ice40/Blackbox.scala#L75

Then you can instanciate those primitives as regular component in your SpinalHDL description

Riconec, 2020-01-03:
Thanks @Dolu1990, I will try that

closed 2020-01-03 by Riconec