SpinalHDL/SpinalHDL #251: [Request]: Remove assigns for direct connections to parent using autoConnect

lsteveol, 2020-01-08:
I wondering if something like this can be adjusted. Similar to https://github.com/SpinalHDL/SpinalHDL/issues/132. If I have a Bundle that I use up a hierarchy, outputs seem to get reassigned.

Here is an example
```scala
class MyBundle extends Bundle{
  val a = in  UInt(8 bits)
  val b = in  UInt(8 bits)
  val c = out UInt(8 bits)
}

class TestBundle extends Bundle{
  val somein  = in  Bool
  val someout = out Bool
}

class MyMod extends Component {
  val io = new MyBundle
  val tb = new TestBundle
  
  io.c := io.a + io.b
  
  tb.someout := False
  
  noIoPrefix()   
}

class MyMod2 extends Component {
  val io = new MyBundle
  val tb = new TestBundle
  
  val mod = new MyMod
  
  mod.io.a := io.a
  mod.io.b := io.b
  
  io.c := io.a + mod.io.c
  
  tb <> mod.tb
  
}
```

This will produce:
```verilog
module MyMod (
      input  [7:0] a,
      input  [7:0] b,
      output [7:0] c,
      input   tb_somein,
      output  tb_someout);
  assign c = (a + b);
  assign tb_someout = 1'b0;
endmodule

module MyMod2 (
      input  [7:0] io_a,
      input  [7:0] io_b,
      output [7:0] io_c,
      input   tb_somein,
      output  tb_someout);
  wire [7:0] mod_1__c;
  wire  mod_1__tb_someout;
  MyMod mod_1_ ( 
    .a(io_a),
    .b(io_b),
    .c(mod_1__c),
    .tb_somein(tb_somein),
    .tb_someout(mod_1__tb_someout) 
  );
  assign io_c = (io_a + mod_1__c);
  assign tb_someout = mod_1__tb_someout;
endmodule
```

It would be ideal if it could do this:
```verilog
module MyMod2 (
      input  [7:0] io_a,
      input  [7:0] io_b,
      output [7:0] io_c,
      input   tb_somein,
      output  tb_someout);
  wire [7:0] mod_1__c;
  MyMod mod_1_ ( 
    .a(io_a),
    .b(io_b),
    .c(mod_1__c),
    .tb_somein(tb_somein),
    .tb_someout(tb_someout ) 
  );
  assign io_c = (io_a + mod_1__c);
endmodule
```

I know this is a byproduct of VHDL, but is there a way to remove this intermediate variable? A reason I'm asking is that in ASIC flows you will often need to set a "dont_touch" on a particular signal. This keeps the synthesis tool from inserting a buffer or other logic. These tools can often handle an assign statement, but I know being able to only have 1 signal name is ideal. I've only seen this on outputs, and not on inputs.

I have dug through the SpinalHDL code some, but I'm pretty fresh with it and Scala in general, so I don't really have much to offer in terms of what I've tried. But I'm willing to try some things out if needed. I'm not sure how much effort this would be.

Thanks!

lsteveol, 2020-01-14:
Hi @Dolu1990 ,

Thanks for the feedback. Would it be possible/advisable to add something in the SpinalConfig (like a subCompOutputBind) that could default to true but be overridden during generation and it select either of the two?

I will try something like this out and submit a fix, might be later this week due to work.

lsteveol, 2020-01-16:
So if I replace with ```referencesOverrides(io) = nameOfTheCorrespondingPinYouWant``` I still seem to get an assign statement.

Could you point me to the area of code where the assigns are created for the outputs and I can see if there is a solution? I noticed there was a NormalizeInputs object, I'm not sure if something like that needs to be done for the outputs as well.

numero-744, 2022-11-14:
Is this issue outdated? @Dolu1990

lsteveol, 2022-11-14:
Sorry, I haven't spent any time with Spinal in a few years. I can close unless someone would like me to keep it open

Dolu1990, 2022-11-15:
It isn't outdated, that behaviour is still there, but i would kind of opt for a no fix for now, as it isn't a bug, but mostly cosmetic.