SpinalHDL/SpinalHDL #252: Disable sbt interactive features in Travis

piegamesde, 2020-01-11:
Currently, the output of the Travis CI when failing is very bloated. There are many empty lines and progress information that dilute the actually interesting stuff. I suspect this is because of "interactive" features that work well in user terminals, but not when the output is written to a file.

I don't know much about sbt, I couldn't find a name for this "feature" or how to disable it.

Example:

```
  | => core / Compile / unmanagedSources / inputFileStamps 0s
[error] Failed tests:
[5 empty lines]
  | => core / Compile / unmanagedSources / inputFileStamps 0s
[error] 	spinal.tester.scalatest.MandelbrotTesterCocotbBoot
[5 empty lines]
  | => core / Compile / unmanagedSources / inputFileStamps 0s
[5 empty lines]
[error] (tester / Test / test) sbt.TestsFailedException: Tests unsuccessful
[error] Total time: 944 s (15:44), completed Jan 11, 2020 9:32:29 PM
```

Dolu1990, 2020-01-14:
To be honnest i don't remember exactly about it.
@1138-4EB Does it tell you something ?

Dolu1990, 2020-01-14:
Ahh maybe that's also due to the SBT update ? it behave quite differently with the console

piegamesde, 2020-01-15:
Try running the CI builds with `sbt --supershell=false`. I've tried `sbt --batch` as well and it seems to work, so maybe `--batch` implies `--supershell=false`?

PR #263 (2020-01-27): Fix sbt supershell in travis
Add "supershell=false" flag like suggested in #252.
I noted that the output is more clear and without unnecessary spaces.
Like you can see in here: https://travis-ci.com/wifasoi/SpinalHDL/jobs/280481629

@Dolu1990 if ok with you I'll merge to dev (or should i merge it to master?)

This will fix #252
closed 2020-01-27 by Dolu1990