SpinalHDL/SpinalHDL #254: Set name of component

piegamesde, 2020-01-13:
I'd like to change the name of my module when it's compiled to Verilog. This is especially relevant when including black box modules, because it must name my class "SOME_CAPITALIZED_BULLSHIT" so that it is instanciated correctly. I'd rather have something like `setModuleName(string)` to override the use of the class name and keep my camel case naming convention internally.

lsteveol, 2020-01-13:
On a Component there is a method ```setDefinitionName``` in which you can pass a string for the name. If you do this on the top level, if will also rename the output Verilog file to the same, with ".v" file extension of course.

```
setDefinitionName("wackawacka")

//And in wackawacka.v

module wackawacka (
      input   apb_clk,
      input   apb_reset,
```

lsteveol, 2020-01-13:
Also, Spinal should keep the name as you declare it on the BlackBox. I have several BBs and we use normal lowercase+underscore (some_module_name) naming convention for 99% of our Verilog. So I will keep that naming convention for the BlackBox class name, but then CamelCase anything that is extends a Spinal Component

piegamesde, 2020-01-13:
I prefer using the Scala naming convention when writing scala. `setDefinitionName` does what I want, thank you. Is this documented somewhere and if not, where should I add it?

lsteveol, 2020-01-13:
I did not see it documented anywhere. I just happened to stumble across it the other week while poking around the code base :) . @Dolu1990 can suggest what to do about adding it to the docs.

Dolu1990, 2020-01-14:
Hi @piegamesde 
Hmm right, if you have the time, i'm happy to merge a doc pull request :)
Here it can be added : 
https://github.com/SpinalHDL/SpinalDoc-RTD/blob/master/source/SpinalHDL/Structuring/components_hierarchy.rst

closed 2020-02-06 by wifasoi