SpinalHDL/SpinalHDL #255: Multi-word driveFlow  stays valid forever

distributed, 2020-01-13:
The `valid` behavior of `BusSlaveFactory.driveFlow` is different for flows that fit within the bus data width and those that don't.

For the former, `valid` is asserted for one clock cycle, with a combinatorial signal.

For larger flows, `valid` is asserted in the next clock cycle and never deasserted.

There is a `regValid` in `driveFlow`:
```
val regValid = Reg(that.valid) init(False)
```

which is set right below

```
onWrite(address + ((wordCount - 1) * wordAddressInc)){ regValid := True }
```

but never reset.

Maybe I misunderstood something, but I believe that `regValid` should stay asserted for only one clock cycle.

Dolu1990, 2020-01-14:
Hooo right, my bad, clearly a thing to fix

distributed, 2020-01-16:
No worries. I am currently not using it, but I stumbled across it while reading the code :)

commit a4b93beb (2020-02-02): Fix #255 by ensuring a single pulse on regValid

Also fixes checking bitOffset for calculating wordCount.
closed 2020-02-02 by Dolu1990