SpinalHDL/SpinalHDL #260: Simulation Diagnostics and Feedback

distributed, 2020-01-17:
Below is what I brought up on Gitter :) The questions asked point to places, where new diagnostics could be printed.

I have set up a number of simulations for a component of mine and I have trouble interpreting the results I am getting.

Inside the component, I have this assertion:

```assert(!recordWriteLock, "push first fire while holding record write lock", FAILURE)```

When I run it, i might get this: (i seem to depend on random signals, so assertions fail randomly as well)

```
[info] [Progress] Verilator compilation done in 2386.726 ms
[info] [Progress] Start RecordFifo basic simulation with seed -7657521877275568118, wave in /home/joe/FPGA/prj/hdl/cisc/./simWorkspace/RecordFifo/basic.vcd
[info] [Done] Simulation done in 26.360 ms
[info] [Progress] Start RecordFifo oneoverflow simulation with seed -7497477908939138281, wave in /home/joe/FPGA/prj/hdl/cisc/./simWorkspace/RecordFifo/oneoverflow.vcd
[info] FAILURE push first fire while holding record write lock
[info] - /home/joe/FPGA/prj/hdl/cisc/tmp/job_1/RecordFifo.v:87: Verilog $finish
[info] FAILURE push first fire while holding record write lock
[info] - /home/joe/FPGA/prj/hdl/cisc/tmp/job_1/RecordFifo.v:87: Verilog $finish
[info] - /home/joe/FPGA/prj/hdl/cisc/tmp/job_1/RecordFifo.v:87: Second verilog $finish, exiting
[success] Total time: 9 s, completed Jan 16, 2020, 10:26:36 AM
```

I have set up 4 different simulations, however here I am only seeing two of them mentioned, basic.vcd and oneoverflow.vcd.

- Can I assume that the failure happened in trace `oneoverflow.vcd`, because it's the last one printed before the error?
- Because it's a `FAILURE`, simulation stops right when the assertion fails. Time is not printed. How would I spot the failed assertion when it's just an `ERROR`?
- Why is the overall process (last line) successful even if the simulation failed?

Dolu1990, 2020-01-30:
Have to fix that, there is no link between verilator error and SpinalSim.

Dolu1990, 2020-02-03:
Got it : 
https://github.com/sifive/verilator/blob/master/include/verilated.cpp#L1489

Dolu1990, 2020-02-03:
I pushed a fix for SpinalSim in the dev branch to capture the assertion from the verilog.

> How would I spot the failed assertion when it's just an ERROR?

Hmm maybe there is a verilator option for that, have to check

> Why is the overall process (last line) successful even if the simulation failed?

What do you mean exactly ?

distributed, 2020-02-03:
> I pushed a fix for SpinalSim in the dev branch to capture the assertion from the verilog.

SBT noob question: How can I try those changes?

>> Why is the overall process (last line) successful even if the simulation failed?

> What do you mean exactly ?

The last lines of the build output says `success` (also in green) even though the simulation failed. I would have expected the build output to say `failure` or somesuch. But maybe this view is not correct, because while running the test bench failed, running the tests was successful?

Dolu1990, 2020-02-03:
I just pushed another fix
So basicaly, if you want to try it out in your project, you can follow : 
https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-commited-on-git

distributed, 2020-02-03:
Getting the new version is a breeze, thanks!

Unfortunately I can't compile with Spinal from the dev branch. I have a custom Bundle with a custom Mux object that connects `ret.write << StreamMux(select, inputs.map(_.write))` where `write` is a Stream. 

```
[error] HIERARCHY VIOLATION : (toplevel/[StreamMux]/io_inputs_2_ready : out Bool) is driven by (toplevel/writeGain/io_trx_write_ready : in Bool), but isn't accessible in the toplevel component.
[error]     spinal.lib.StreamMux$.apply(Stream.scala:659)
[error]     spinal.lib.StreamMux$.apply(Stream.scala:654)
```

There are other `io_write_trx_ready` signals reported where I cannot see where I would drive the `ready`. This code compiles with 1.3.8. I saw that there are changes between 1.3.8 and dev regarding `StreamMux`. Do you have a guess what could have gone wrong?

Dolu1990, 2020-02-03:
I found something, and that was very likely the issue you had. So it should be fixed now : 
https://github.com/SpinalHDL/SpinalHDL/commit/384858e3307dedb0f48ec1b4c4d21a93a69668b3#diff-43fb6fda1ee53e4a12ed702184e9a8a3R659

Let's me know how it goes :)

distributed, 2020-02-04:
It works!

Output on failure now looks like this:

```
[info] [Error] Simulation failed at time=107488
[error] Exception in thread "main" spinal.sim.SimFailure: Verilog assertion failure
[error]         at spinal.sim.SimManager.runWhile(SimManager.scala:229)
[error] isc / Coat spinal.sim.SimManager.runAll(SimManager.scala:190)
[error] 	at spinal.core.sim.SimCompiled.doSimPostSeed(SimBootstraps.scala:227)
[error] 	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:203)
[error] 	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:200)
[error] 	at CiscSim$.main(cisc.scala:994)
[error] 	at CiscSim.main(cisc.scala)
[info] FAILURE timeout data counter overflow
[info] - /home/joe/FPGA/proj/hdl/cisc/tmp/job_1/Cisc.v:2155: Verilog $finish
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 14 s, completed Feb 4, 2020, 9:52:48 AM
```

File:Linenumber is not printed for the assertion in Spinal source. However it's easy to spot the place where the assertion failed according to a sufficiently unique assert message.

Very nice, thanks!

Dolu1990, 2020-02-05:
All good right ?

closed 2020-02-05 by Dolu1990

distributed, 2020-02-06:
Yes, thanks!