SpinalHDL/SpinalHDL #264: AXI resize address bus

piegamesde, 2020-01-29:
I have the simple problem, that I want to connect my 12 bit address space AXI slave to a 32bit AXI master. The wires can simply be `resized` without any further logic, since I own the whole address space.

The only solution I found is to manually connect every wire, resizing the address bits when needed. But I'd prefer to have some kind of transforming method that I can call and then use `<>` to autoconnect.

Dolu1990, 2020-01-29:
Did you tried the myMaster >> mySlave operator ?

piegamesde, 2020-01-29:
Never thought of this, I honestly don't really know the difference between all these operators (are they documented somewhere? ^^)

It does seem to work with different bit widths, but it now complains because the master has more pins than the slave (the master has all AXI extensions, the slave only a few). Is there a similar operator that connects what is there, and leaves the rest unconnected?

piegamesde, 2020-02-02:
@Dolu1990 *ping*

Dolu1990, 2020-02-02:
Basicaly, the >> operator can substitut missing signal which have default values, but if your master side has the signal, let's say aw_burst, then the slave has to implement it, else you have to manualy wrap things together :/

Another way is to use the assignSomeByName, and patch things afterward

Dolu1990, 2020-02-02:
Else we could add a lazy connection operator which connect things what ever the assumption it take ?

piegamesde, 2020-02-02:
This is unfortunate. There should be a way to connect in a way that the input can be ignored (leaving the outputs of the other side unconnected). Is it reasonable to change the `>>` operator for Axi to allow the Master to have a wider config or should we prefer a more general solution? `assignSomeByName` solves the missing wires issue, but it can't handle different directions.

Dolu1990, 2020-02-02:
Sorry, i'm not sure what a input / output mean as it can be thinked the other way around, can you precise ? let's say master input/output    slave input/output

piegamesde, 2020-02-02:
@Dolu1990 I've updated my comment

Dolu1990, 2020-02-02:
> (leaving the master/outputs of the other side unconnected

Normaly that's already implemented in the >> operator

>  Is it reasonable to change the >> operator for Axi to allow the Master to have a wider config

That's should be already ok for the address

Which is exactly your case, (config on both side)

Dolu1990, 2020-02-02:
> (leaving the master/outputs of the other side unconnected
> Normaly that's already implemented in the >> operator

Hmm no sorry i'm wrong, basicaly this isn't implemted, as that's not a safe connection, as maybe the master need the feature created by his output pin.

lsteveol, 2020-02-03:
I'm not sure I follow the entire issue here, but just as an additional point of reference, I have a proprietary interface that can be used in multiple configurations. I wanted to be able to easily connect these two interfaces without explicit connections on each instance inside a component. `assignByName` and `assignSomeByName` weren't clean workarounds for me.

What I ended up settling on was creating a "connection method" which I define in the `Bundle`. It would look something like this

```scala
class MyBundle extends Bundle{
  //my io
  var/val someVariableIMayUseLater = 0

  def connectIntf(intf : MyOtherBundleType){
    //connections between two bundles with similar but not the same interfaces
  }
  def connectIntf(intf : MyBundle){
    //connections between two bundles that are the same, but maybe have different widths
    //or I need to tieoff/connect differently
  }
}
```

I have a case where based on the config of the design, I need to tie off a portion of the MyBundle signals but still connect the remaining ones to the complementary bundle. This allowed me to easily do that.

Obviously something like `<>` would be more ideal, but this still cleans up the code where the Components/Bundles are instantiated, especially if I have multiple instances of each.

There may be better ways to do this, however this has worked for me.

Dolu1990, 2020-02-05:
> I have a proprietary interface 

Ahh i was thinking that the issue was specific to AXI

> assignByName and assignSomeByName weren't clean workarounds for me.

Right, i agree

> I have a case where based on the config of the design, I need to tie off a portion of the MyBundle signals but still connect the remaining ones to the complementary bundle. This allowed me to easily do that.

Hmm i'm not sure what is the issue.

```scala
  def connectIntf(intf : MyOtherBundleType){
    if(intf.config.x) this.a := intf.a
    if(intf.config.y) this.b := intf.b
  }
```

So using the scala if statement as a elaboration tool to connect things conditionaly

Does that kind of anwer the question ?

Could you bring a practical case if not ?

lsteveol, 2020-02-05:
Hi @Dolu1990 

My comments were mainly meant as a potential workaround for @piegamesde . I believe his issue is AXI related,but I wasn't exactly following his issue.

The Scala `if` statement would work if the configs were limited in number and/or only a few signals. For me the issue was the interface is common for one IP (let's call it IP A with `MyBundle`), and depending on the connected IP (let's call them B through F, with `MyOtherBundleType[B-F]`) the connections are all different. The IP's B through F also have different interfaces, so we can't *easily* use the same `Bundle`. A certain signal on IP A may be used for B, D, and F, but not the others. So you start to have this large matrix of what is or isn't connected. And this is the case for a large number of signals.

I can't really explain much more for proprietary reasons.

So I create as many `Bundles` as needed for each IP, then just write a connection rule for each type. While not the best looking, it is fairly explicit and is still easier than traditional verilog connections.

Spinal/Scala have enough support for this, so I don't think there needs to be any modifications made. I was just offering a *potential* solution :)

Dolu1990, 2020-02-05:
@Isteveol Hoo sorry, my bad, i didn't noticed it was a different acount XD

piegamesde, 2020-02-08:
I just ran into a similar issue with connecting stuff: How an I leave an AXI master or slave unconnected (dangling)? I just want to compile my current WIP design.

lsteveol, 2020-02-08:
If you just need to try something for compilation, you can simply tieoff the input signals directly and leaving the outputs dangling. Just tie the respective valids/readys to 0/False and don't try to access anything on the bus.

piegamesde, 2020-02-08:
Yes, but how can I automatically drive all inputs to 0 while leaving the outputs unconnected?

piegamesde, 2020-02-11:
Yet another similar issue: How can I use a lite master to drive a full slave? The specification says it works but I don't know how to tell Spinal that it's okay that one has more elements than the other.

numero-744, 2022-11-22:
Inside a component you can assign all outputs to 0 with:

```scala
  this.getAllIo.foreach { io => if (io.isOutput) io := io.getZero }
```

For the other things, I don't know if those have been implemented since.

If they have not and if you still need them, please feel free to open issues.

closed 2022-11-22 by numero-744