SpinalHDL/SpinalHDL #266: Merge `AXILite4` into `AXI4`

piegamesde, 2020-02-05:
AXI4 and it's lite variants are not two distinct specifications. In reality, it is more like AXI is a set of features, where AXI lite uses less of them. Therefore, I find it quite cumbersome to model them using different classes. For example:

- I only need the `id` extension to connect a lite slave to a full master. But I'd then have to replace all types with the equivalents of the other package
- When writing converters for different address or data widths, the lite version is essentially just a subset of the original one, resulting in unnecessary code duplication.

I propose to deprecate the lite classes and to enhance `AxiConfig` to better handle lite masters/slaves.

Dolu1990, 2020-02-08:
From an interconnect perspective, the lack of ID can be a big issue, as the ID is specificaly extended by the interconnect to store the information of where things should be routed back.

About AXI4-lite, is it allowed to do unaligned accesses ?

What is missing in the AxiConfig to be able to model AxiLite ?

piegamesde, 2020-02-08:
All accesses are aligned; unaligned accesses are modeled through the use of strobes. See section `B1.1.3 Write strobes` of the specification. A slave may choose to:

- Implement strobes and use them
    - This is mandatory for slaves that provide memory access
- Ignore strobes and always use the full bus
- Error out on a case-by-case basis
    - This allows a slave to implement only subsets of strobes

piegamesde, 2020-02-10:
> What is missing in the AxiConfig to be able to model AxiLite ?

I think the only thing missing in `AxiConfig` is some configuration presets for lite masters/slaves.

Dolu1990, 2020-02-11:
Ahh so,

To me, AXI and AXI-lite are two different things, which require different approches in adapters/interconnect design.
But i have no concern about adding a AxiLite like config function which take addressWidth/dataWidth int eh AxiConfig companion object. Plus we could add a field in the AxiConfig class to specify if the bus follow the Axi or AxiLite rules that you described two messages ago.

Sound good ?

piegamesde, 2020-02-11:
Sounds good. Additionally, there should be some easy AxiLite -> AxiFull bundle conversion mechanism to reduce the overhead for upgrading an implementation (I most often start with lite and then implement bursts if performance becomes an issue)

Readon, 2023-01-04:
It seems the existing facility is not easy to reduce the Axi4 to AxiLite4, so should we label this as wontfix?

numero-744, 2023-01-04:
Maybe rename it to "Add conversions `AXILite4` <-> `AXI4`" and wait for #1002 to be merged?

numero-744, 2023-01-04:
Indeed, but the discussion above shows an evolution of the requirements from the issue author from merging them to implementing conversions. But you are right, this is the quick solution, maybe not the cleaner. Another approach can be to close this issue and open another one which mentions this one (and ping the author or mention the new one here too)?

Readon, 2023-01-04:
> Indeed, but the discussion above shows an evolution of the requirements from the issue author from merging them to implementing conversions. But you are right, this is the quick solution, maybe not [...]

After you mentioned #1002 , I think this issue can be closed.

closed 2023-02-10 by Readon