SpinalHDL/SpinalHDL #268: FSM does not compile

piegamesde, 2020-02-10:
I have a really simple FSM:

```scala
  val fsm: StateMachine = new StateMachine {
    val idle = new State with EntryPoint
  }
  fsm.build()
```

And I get:

```
********************************************************************************
********************************************************************************

Reserved name fsm_stateReg is not free for (toplevel/fsm_stateReg :  SpinalEnumCraft)
********************************************************************************
********************************************************************************

Reserved name fsm_stateNext is not free for (toplevel/fsm_stateNext :  SpinalEnumCraft)
********************************************************************************
********************************************************************************
```

Dolu1990, 2020-02-10:
You should not manualy call the fsm.build(), as that's already registred to be automaticaly call when the component elaboration is done

piegamesde, 2020-02-10:
@Dolu1990 But if I leave out build then `fsm.stateReg == null`. How can I access the fsm's state instead?

Dolu1990, 2020-02-10:
What do you mean by accessing, knowing if the state is active ? fsm.isActive(idle)

piegamesde, 2020-02-10:
@Dolu1990 I want to expose a number representing that state (to put it into a debug register I map into memory)

Dolu1990, 2020-03-18:
You could access the enumeration state if you extend the build function of the state machine. There you have stateReg.

But all of that is a bit tricky :/

piegamesde, 2020-03-18:
One the one hand, accessing the state of a FSM is really helpful for debugging. On the other hand, the encoding of the states may be changed and optimized away in any form, so relying on this would break that optimization.

I worked around it by assigning a wire a different value when each state is active. We could add some helper methods to do this but it's probably not worth the cost.

closed 2020-03-18 by piegamesde