SpinalHDL/SpinalHDL #275: `setAllBits` does not work with `asBits`: NO DRIVER ON value

piegamesde, 2020-02-28:
This simple example instantiates a sub-component and sets all its input wires to `True`.

```scala
object Test {
  def main(args: Array[String]) {
    val config = SpinalConfig(
      mode = Verilog)
      .generate(new Component {
        val o = out UInt (8 bits) // Dummy output against pruning

        val mySubComponent = new Component {
          val i = in UInt (8 bits) // The relevant input
          val o = out UInt (8 bits)
          o := U(i.asBits) & 0xF // Dummy logic to avoid signal pruning
        }

        mySubComponent.i.asBits.setAllTo(True) // Does not work
        o := mySubComponent.o
      })
      .printPruned()
      .printPrunedIo()
      .printUnused()
  }
}
```

But this doesn't work, when compiling I get `NO DRIVER ON (toplevel/mySubComponent/i : in UInt[8 bits])`. The interesting part is, when I call `setAllBits` directly without converting to `asBits`, it works.

Dolu1990, 2020-02-29:
Hi,

That's normal, it is by design, basicaly : 

mySubComponent.i.asBits  
As bits return a new signal which is assigned by by mySubComponent.i
So, setAllTo(True)  will not set mySubComponent.i but the new casted signal

Sound fair ?

piegamesde, 2020-02-29:
If `asBits` returns a new signal which is assigned by the original value, and I then go and assign that signal a new value with `setAllTo`, isn't this double assignment? Shouldn't this trigger some compiler error then?

Does this mean that I can never modify a signal's `asBits` representation, not only in this case? Does this mean that `mySubComponent.i.asBits := B(0xA1)` won't work as well?

What is the correct way to achieve this, then?

Dolu1990, 2020-02-29:
>  isn't this double assignment? Shouldn't this trigger some compiler error then?

Likely SpinalHDL check first for unasigned things, then for double assignement, that's probably why

> mySubComponent.i.asBits := B(0xA1) won't work as well

Right, will not work.
But all Data types have a assignFromBits(yourBits) method, this is the way to go to load raw stuff into a data structure. Is that what you wanted to do ?

piegamesde, 2020-02-29:
> Right, will not work.
I think some compiler feedback when writing such things would be really useful.

Currently, I work around the issue by using `.assignFromBits(Mux(cond, S(-1, 8 bits).asBits, S(0, 8 bits).asBits))`, but that doesn't read nearly as well as `.setAllTo(cond)`.

closed 2020-04-09 by piegamesde

nanoric, 2023-08-25:
> Hi,
> 
> That's normal, it is by design, basicaly :
> 
> mySubComponent.i.asBits As bits return a new signal which is assigned by by mySubComponent.i So, setAllTo(True) will not set mySubComponent.i but the new casted signal
> 
> Sound fair ?

I'm curious if this is "by design" due to some kind of limitation, or to reduce some common mistakes?

Dolu1990, 2023-08-27:
It is by design, would be technicaly fesible. asBits return a new hardware signal instead of a "proxy". Currently the only cases where you get proxy is for BitVector.apply and Vec(x, y z)