SpinalHDL/SpinalHDL #278: Failed to Simulate BlackBox Modules contains $error()

depctg, 2020-03-03:
When I tries to start simulating a black box module contains `$assert` and `$error`, I got the following error:

```bash
Undefined symbols for architecture x86_64:
  "sc_time_stamp()", referenced from:
      Vunamed::_initial__TOP__1(Vunamed__Syms*) in Vunamed__ALL.a(Vunamed__ALLcls.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Vunamed] Error 1
Exception in thread "main" java.lang.AssertionError: assertion failed: Verilator C++ model compilation failed
```

When I comment out the lines contains $error, the error message is gone. I guess this is caused by Verilator. It wants to attach timestamp to the error message, but it relies on cpp function `sc_time_stamp`. I see some related messages in [verilator repo](https://github.com/grg/verilator/blob/9caffe330b602b630a541207dbb6606e6e3be183/bin/verilator#L3557). I guess theres already java side hooks but external rtls relies on cpp ones.

Zhiyuan

Dolu1990, 2020-03-13:
Hmm so, i should add in the C++ main generated by SpinalHDL a implementation of sc_time_stamp ?

Dolu1990, 2020-03-13:
Does sc_time_stamp give the simulation time ? or real time ? Because if that's simulation time, that's quite enoying as it do not give any context, so not realy compatible with multiple simultion running at the same time, unless using thread context variables

depctg, 2020-03-14:
Thanks for your reply! I guess we need to generate a cpp `sc_time_stamp` function. Currently I do not relies on time information provided by these verilog $errors, so an empty functions to make the compiler happy should be good enough to me.

And, Unfortunately, `sc_time_stamp` returns simulation time. You can check the [example implantation](https://github.com/grg/verilator/blob/9caffe330b602b630a541207dbb6606e6e3be183/bin/verilator#L1580).

Dolu1990, 2020-03-14:
@depctg I pushed something in dev which should fix it, could you give a try ? Do you know how to try unreleased SpinalHDL ?
https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-commited-on-git

closed 2020-03-18 by Dolu1990