SpinalHDL/SpinalHDL #281: Specifying Seed for the Simulation

distributed, 2020-03-07:
I have a simualtion that failed once and printed this:

```
[info] [Progress] Start JpegWriter jpegwriter simulation with seed 8223987900762028738, wave in /home/joe/FPGA/proj/hdl/cisc/./simWorkspace/JpegWriter/jpegwriter.vcd
```

Now I am trying to rerun the same simulation, like so:

```
compiledSim.doSimUntilVoid("jpegwriter", seed=8223987900762028738L) { dut =>
```

However now the simulations runs with a different seed:

```
[info] [Progress] Start JpegWriter jpegwriter simulation with seed 3313447782858668379, wave in /home/joe/FPGA/proj/hdl/cisc/./simWorkspace/JpegWriter/jpegwriter.vcd
```

However, a seed a given to `doSim` seems to result in always the same (different from the original) seed.

Dolu1990, 2020-03-07:
Fixed in dev ^^
Thanks :)

closed 2020-03-07 by Dolu1990

distributed, 2020-03-09:
Merci beaucoup! I will start using it tomorrow :)

distributed, 2020-03-10:
Works like a charm.