SpinalHDL/SpinalHDL #284: Reproducible builds

piegamesde, 2020-03-12:
The generated Verilog code contains a header like this:

```Verilog
// Generator : SpinalHDL v1.3.9    git head : 0f14fcc31e1099ab7fb106009a605d0d6e7be21a
// Date      : 12/03/2020, 23:01:57
// Component : SlowdownTest
```

Including the compile date is generally a bad idea, because it not reproducible by definition.

Dolu1990, 2020-03-12:
Ahhhhh that date stuff is quite usefull XD

But i agree, it make sense. Probabily we should disable it by default, and add an option to get the date ?

piegamesde, 2020-03-12:
Yes, please.

jijingg, 2020-03-20:
does it better to set `headerWithDate = true` as default ,  the date is usefull , and do nothing bad , for other pepole they will recognize the generated verilog version at first sight by date.

piegamesde, 2020-03-20:
No, this should not be on by default, because the packager often is not the same person as the one writing the hardware code. If only the compiled Verilog is available (which sadly is not that uncommon), there is no way to recompile with the changed setting and patching all files would be the only work-around.

But maybe there are other information that are similarly helpful but don't break reproducible builds? Version information (or the Git commit) are okay for example, since you'd pin all dependencies anyways to get a reproducible build.

Dolu1990, 2020-03-24:
@piegamesde 

> there is no way to recompile with the changed setting and patching all files would be the only work-around.

Sorry, i'm not sure to understand the context, can you explain more ?

> Version information (or the Git commit)

That's a good idea

piegamesde, 2020-03-24:
> Sorry, i'm not sure to understand the context, can you explain more ?

I just noticed that my example is pretty pointless, so please ignore that ^^.

Dolu1990, 2020-03-24:
@piegamesde @jijingg  i added the git hash in the generated RTL, enabled by default

Also, i reopen the issue, as the subject is still open to me

typingArtist, 2020-04-18:
I would also consider the Spinal version plus hash as something that can break consistent builds. I am open to the default state (on or off), but we should definitely have an option of disabling this in the output as well.

piegamesde, 2020-04-18:
@typingArtist I went from the assumption that if you care about build reproducibility, you'll have to pin all the software dependencies and build tools anyway. What is your use case that breaks this assumption?

typingArtist, 2020-04-19:
@piegamesde There are different aspects to reproducibility. Most of the time you want to keep all aspects stable and only change one aspect, say, the time of compilation/build. Another time you want to change only Scala (e.g. 2.11 to 2.12, or 2.12.4 to 2.12.11, or alike). Again you would like to monitor/compare the same output. Boom, the hash of the Spinal Version changes.

Furthermore, it is questionable what this line of version information actually covers. The SpinalHDL core version? The SpinalHDL lib version? And what about all other libs not covered in this version string?

I don’t want to rant here or sound like someone who wants to kill that line altogether, but it appears to be »an EDA thing« to put dates and tool versions into each and every (intermediate) output. I think that, at least in vital outputs it should be optional to have this information in.

piegamesde, 2020-04-19:
@typingArtist Okay, so I'd say what you describe is less about "reproducibility" than about "stability" and I understand your point. The hash was just an idea to replace the time stamp while maintaining reproducibility, but yes, it should be optional (and opt-in) as well.

piegamesde, 2021-06-16:
I'd like to bring this up again. I propose two new ways to deal with time stamps:

1. Don't use the current time, but the modification time of the Scala source file(s)
2. [`SROUCE_DATE_EPOCH`](https://reproducible-builds.org/docs/source-date-epoch/) is a standardized environment variable with defined semantics for this exact purpose. Build environments that ought to be deterministic can set it. It is kind of a last-resort method, but it is good to have this implemented nevertheless.

Dolu1990, 2021-06-18:
This sound good to me :)

numero-744, 2022-11-22:
Hmm [SOURCE_DATE_EPOCH](https://reproducible-builds.org/docs/source-date-epoch/) is not set by `sbt`

```scala
object TestEnv extends App {
  println(
    sys.env
      .get("SOURCE_DATE_EPOCH")
      .map(sde => new java.util.Date(sde.toLong * 1000))
  )
}
```

andreasWallner, 2022-11-30:
@numero-744 isn't `SOURCE_DATE_EPOCH` supposed to be set by the distribution (in the namings of reproducible-builds.org) to be consumed by sbt, rather than being set by sbt? You basically overwrite the real current time by setting that environment variable, especially for multi-stage builds?

numero-744, 2022-12-01:
Ah, so we could release the option and let the user manage the environment variable as an input?

commit c670e170 (2020-03-13): Fix #284, date in header is new disabled by default