SpinalHDL/SpinalHDL #285: Fix point operations

piegamesde, 2020-03-14:
I'my currently trying out fix point operations and I'm missing a few features:

- Widen a fix point number (both the exponent and the resolution, separately)
- Cast between signed and signed (The doc says that there is a `toSFix`/`toUFix` method, but the compiler says there isn't: `value toSFix is not a member of spinal.core.UFix`)
- Sign operations (extract sign, abs, negate)

jijingg, 2020-03-18:
Maybe you can use UInt SInt directly, the UInt/SInt  already supports most fixpoint operations. see #233 
>Widen a fix point number

```scala 
val a = SInt(8 bits)
val b = a.fixTo(10 downto 0 ) //will auto expand sign bit 
val b = a.tag(SQ(8,4)).fixTo(SQ(10,4))
//also support cut operation 
val c = a.tag(SQ(8,4).fixTo(SQ(6,2), RoundType.ROUNDTOINF, sym = true)  
val c = a.fixTo(8 downto 2) 
```
> Cast between signed and signed 

I may not understand, Is this what you want? 
```
val a ,b = SInt(8 bits)
a.asBits ## b.asBits
```

> Sign operations

```
val a = SInt(8 bits)
a.sign 
a.abs 
a.absWithSym
- a
```

wswslzp, 2020-05-22:
Why not directly add fixTo() function as a member of UFix/SFix?

numero-744, 2022-10-27:
Hmm this issue can be closed, right?

Dolu1990, 2022-10-31:
I would say, if possible, efforts should be made toward the recently introduced AFix class.

But utilities in a PR are always welcome.

closed 2022-10-31 by numero-744