SpinalHDL/SpinalHDL #289: Issue with initialized memory (Mem)

saahm, 2020-04-02:
This is probably related to issue #160 -- might even be moved into that issue if it turns out to be directly related.
I initialize a memory directly via a function in a Tools class -- pretty much as proposed as current work-around by issue #160 
```scala
  def readBytesFromTxt(path: String): Array[BigInt] = {
    val buffer = new ArrayBuffer[BigInt]
    for(line <- Source.fromFile(path).getLines){
      val bytes : Array[String] = line.split(" ")
      val parsedWord : BigInt = 0
      for(byte <- bytes){
        Integer.parseUnsignedInt(byte)
      }
    }
    buffer.toArray
  }
```
And inside my Memory wrapper with 
```scala
mem.initialContent = Tools.readmemh(initFile)
```
It generates verilog that loads the generated file via $readmemb into the designated register (which SpinalHDL puts into that .bin file).
But neither in SpinalHDL executed simulation, nor in a handwritten Verilog testbench it seems that the memory gets initialized non-zero.

Any ideas on how I can look further into how to get that going?

Dolu1990, 2020-04-03:
Hi, 

That's weird. Could you send me a minimal exemple for me to run it and hit the issue ? (Minimal sbt project, or just the scala + initFile)

Thanks :)

saahm, 2020-04-04:
I've thrown a running example into https://github.com/saahm/MemoryIssue 
Sorry for not simplify it further than that. 
Its basically an RTL exercise doing nothing of use at the moment but fetching numbers from ram to calculate stuff. These i wanted to load via a hexfile in 32 bit chunks which get interpreted as 2 16-bit unsigned integers.
Maybe you see what I do wrong in the wrapper for memory already (https://github.com/saahm/MemoryIssue/blob/master/src/main/scala/issue/Memory.scala). Notice I resized the data/address so I can run the simulation in the first place. So either the memory wrapper is being built wrong or the reading and initializing of the hexfile might be. 
I'm sure it might end up being my usage of the SpinalHDL features not correctly

Dolu1990, 2020-04-06:
@saahm I checked it, so there is some usages mistake, but anyway i need to add more checks to catch such cases.

So, to fix things on your side : 

- class Memory(memoryWidth : Int
- val mem = new Mem(Bits(memoryWidth bits), wordCount)
- io.sb.SBrdata := mem.readSync(

I will look on my side to catch more user errors ^^

Thanks for the MemoryIssue git :)

closed 2020-04-06 by Dolu1990

saahm, 2020-04-06:
I think I found my issue. As suspected, my usage of the features was not correctly. I have updated the Repository for future reference. 
I can now load 32 bit values in the 512 word deep ram as init values. 

It was not very obvious to me through the documentation, how and what is and is not allowed to infer certain types of generated HDL code. 

As you mentioned you add these checks in the future. 

Thanks for helping!

Dolu1990, 2020-04-06:
I understand, no worries ^^
Best is to experiement and look for exisiting good to understand the underligning rules, basicaly, you should not use SpinalHDL data types as parametrization, but instead use HardType, maybe check out the StreamFifo implementation  : 
https://github.com/SpinalHDL/SpinalHDL/blob/955358a2270fc77aba1e7c13a1652ebe316775cc/lib/src/main/scala/spinal/lib/Stream.scala#L834

But that example is also using other abstract concepts, carefull ^^