SpinalHDL/SpinalHDL #291: Write to memory with condition on address

saahm, 2020-04-10:
Hey,

I'm trying to implement a Register File for a RISC-V, yet the design generated does not equal the described design in SpinalHDL:

```scala
class RV32RegisterFile(addressWidth : Int, dataWidth : Int, wordCount : Int) extends Component{
  val io = new Bundle{
    val rs1 = in UInt(addressWidth bits)
    val rs2 = in UInt(addressWidth bits)
    val rs1Data = out Bits(dataWidth bits)
    val rs2Data = out Bits(dataWidth bits)
    val wrEna = in Bool
    val rd = in UInt(addressWidth bits)
    val rdData = in Bits(dataWidth bits)
  }
  val regFile = new Mem(Bits(dataWidth bits),wordCount)
  // initialize all registers in register file with 32bit 0 values
  regFile.init(List.fill(wordCount)(B(0,dataWidth bits)))

  io.rs1Data := regFile.readAsync(io.rs1)
  io.rs2Data := regFile.readAsync(io.rs2)

  // this acutally does not translate to hardware disabling write @ address 0
  when(io.rd =/= 0){
    regFile.write(io.rd,io.rdData,io.wrEna)
  }
}
```
generated the following verilog description when used like `val regs = RV32RegisterFile(5,32,32)`:

```verilog
module RV32RegisterFile (
      input  [4:0] io_rs1,
      input  [4:0] io_rs2,
      output [31:0] io_rs1Data,
      output [31:0] io_rs2Data,
      input   io_wrEna,
      input  [4:0] io_rd,
      input  [31:0] io_rdData,
      input   clk,
      input   reset);
  wire [31:0] _zz_1_;
  wire [31:0] _zz_2_;
  reg [31:0] regFile [0:31];
  initial begin
    $readmemb("Top.v_toplevel_cpu_regs_regFile.bin",regFile);
  end
  assign _zz_1_ = regFile[io_rs1];
  assign _zz_2_ = regFile[io_rs2];
  always @ (posedge clk) begin
    if(io_wrEna) begin
      regFile[io_rd] <= io_rdData;
    end
  end

  assign io_rs1Data = _zz_1_;
  assign io_rs2Data = _zz_2_;
endmodule
```
Which does not prohibit writing to address 0 anymore.  

Is there a workaround to force SpinalHDL to generate the condition for the address properly? Or is there a way of generating the condition I have overseen?

Dolu1990, 2020-04-10:
@saahm Hi,

So the Mem.write function has kind of two mode, you can either use it with a explicit write condition, or instead, use the conditional scope in which it is, but not both at the same time.

If you want it to work, you can either do 

```scala
  when(io.rd =/= 0 && io.wrEna){
    regFile.write(io.rd,io.rdData)
  }
```
Or
```scala
regFile.write(io.rd,io.rdData, io.rd =/= 0 && io.wrEna)
```

The reason of this behaviour is to allow the write function to be called by an abstraction, even if that's called in a hardware conditional scope, by explicitly giving the write condition.

Make sense ?

saahm, 2020-04-10:
Thanks for the fast answer. 
The rationale you give doesnt fully make sense to me. This would be good to have in the documentation (at least the code examples with a note about conditions), as I dont see anything about that in there yet.

Dolu1990, 2020-04-21:
@saahm 

Basicaly, imagin you have a function somwhere as : 

```scala

val rom = Mem(Bits(10 bits), 32)
def miaou(address : UInt) : Bits ={
   val counter = Reg(UInt(8 bits))
   counter := counter + 1
   return rom.readSync(address, enable=counter === 42)
}
```` 

And you use it somewhere in your code as a value provider  : 

````scala
when(cond){
   x := miaou(54)
}
```

ok the case make no sense at all, but here the question is, should the rom in the miaou function be sensitive the cond, or not at all, and only follow the counter === 42 condition ?

That's kind of a gray case. in SpinalHDL, you can enforce ignoring the conditional scope (so ignoring the when(cond) for the readSync by providing the ram read enable explicitly (counter === 42)

About the doc, there is a few words about it : 
https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Sequential%20logic/memory.html#syntax

"""
Synchronous write with an optional mask. If no enable is specified, it’s automatically inferred from the conditional scope where this function is called
"""

But i would agree that the explenation isn't realy explicit about it and could be more direct.

saahm, 2020-04-22:
Ah, okay. This kind of usage makes a bit more sense to me. 
About the doc: once one knows what is meant with conditional scope it becomes more clear. 

Maybe something like that added to the doc could help.
```
"""
When enable signals are used inside conditional blocks like `when` for calls on `Mem` only the enable signal will be generated as the access condition. 
"""

//maybe some example as clarification
val rom = Mem(Bits(10 bits), 32)
when(cond){
  io.rdata := rom.readSync(io.addr, io.rdEna)
}

"""
This example will not regard `cond` for accessing the reading on the rom instance.
However, if the intention is to use `cond` then:
"""

io.rdata := rom.readSync(io.addr, io.rdEna & cond)
```

closed 2020-04-22 by saahm