SpinalHDL/SpinalHDL #294: Non descriptive  HIERARCHY VIOLATION  for unrelated errors

azaparov, 2020-04-10:
Code below generate verilog file successfully. However seems to be reporting a lot of hierarchy violation messages when line 127 is un-commented.

```package mylib

import spinal.core._
import spinal.lib._
import spinal.lib.io._
import spinal.lib.fsm._

import scala.util.Random
import java.security.KeyStore.TrustedCertificateEntry

class PCIBus extends Bundle {
  val AD_OUT = UInt(32 bits)
  val AD_IN = UInt(32 bits)
  val AD_OE = UInt(32 bits)
  val CBE   = Bits(4 bits)
  val IDSEL     = Bool
  val FRAME_INV = Bool
  val IRDY      = Bool
  val TRDY      = Bool

  def asSlave(): this.type = { 
    in(AD_IN, CBE, FRAME_INV, IRDY, IDSEL)
    out(AD_OUT, AD_OE, TRDY)
    this
  }
}

object PCIBus {
  val FRAME_ACTIVE = False
  val IDRY_INACTIVE = True
  val IDRY_ACTIVE = False
  val TRDY_INACTIVE = True 
  val TRDY_ACTIVE = False
  val IDSEL_ACTIVE = True

  val CMD_MEMORY_READ  = B"0110"
  val CMD_MEMORY_WRITE = B"0111"
  val CMD_CONFIG_READ  = B"1010"
  val CMD_CONFIG_WRITE = B"1011"
}

//Hardware definition
class MyTopLevel extends Component {
  val io = new Bundle {
    val cond0 = in  Bool
    val cond1 = in  Bool
    val flag  = out Bool

    val state = out UInt(8 bits)
    val pci   = new PCIBus().asSlave()
  }

  val dataMem = Mem(UInt(32 bits), 4)
  // cfgMem.generateAsBlackBox()

  val cfgMem = Mem(UInt(32 bits), initialContent = Array[UInt](1, //devid vendorid
  0,  //status, command
  (0x28000 << 8) | (1), //class code
  0, //header type //standard PCI/PCMCIA header
  0, //bar 0
  0, //bar 1
  0, //bar 2
  0, //bar 3
  0, //bar 4
  0, //bar 5
  0, //cardbus pointer
  (1 << 16) | (0) //subsystem ID, subsystem vendor ID
  ))

  io.pci.TRDY := PCIBus.IDRY_INACTIVE
  //io.pci.AD_OUT := 0 //(default -> 0)
  io.pci.AD_OE := 0 //(default -> 0)
  //cfgMem.write(U"2'1", 1)
  //io.pci.AD_OUT := cfgMem.readAsync(U"4'0")

  val address    = Reg(UInt(32 bits))
  val cmd        = Reg(Bits(4 bits))
  val cfgMemAddr = Reg(UInt(4 bits))  
  val blkMemAddr = Reg(UInt(10 bits))
  val blkMemData = Reg(UInt(32 bits)).init(0)

  val pciFSM = new StateMachine {
    val stateIdle      = new State with EntryPoint
    val stateDecodeCmd = new State
    val stateCfgRead   = new State
    val stateCfgWrite  = new State
    val stateMemRead   = new State

    stateIdle.
      whenIsActive{
        when(io.pci.FRAME_INV === PCIBus.FRAME_ACTIVE) {
          address := io.pci.AD_IN
          cmd     := io.pci.CBE
          
          //handle config read
          when(io.pci.IDSEL === PCIBus.IDSEL_ACTIVE) {
            when(io.pci.CBE === PCIBus.CMD_CONFIG_READ) {
              goto(stateCfgRead)
            }
            when(io.pci.CBE === PCIBus.CMD_CONFIG_WRITE) {
              goto(stateCfgWrite)
            }
          }.otherwise {            
            when(io.pci.CBE === PCIBus.CMD_MEMORY_READ) {
              goto(stateMemRead)
            }
            //goto(stateDecodeCmd)
          }
          
        }        
      }
    stateCfgRead.
      whenIsActive{
        //io.pci.AD_OUT := cfgMem.readAsync(address(3 downto 0))
        cfgMemAddr := address(3 downto 0)
        when(io.pci.IRDY === PCIBus.IDRY_ACTIVE) {
          goto(stateIdle)
        } 
      }
    stateCfgWrite.
      whenIsActive{
        goto(stateIdle)        
      }
    stateMemRead.
      whenIsActive{
        blkMemAddr := address(9 downto 0)
//        blkMemData := dataMem.readAsync(address(9 downto 0))
      }
    stateDecodeCmd.
      whenIsActive{    
        // switch(cmd){
        //     is(PCIBus.CMD_MEMORY_READ) {
        //       goto(stateIdle)
        //     }
        //     is(PCIBus.CMD_MEMORY_WRITE) {
        //       goto(stateIdle)
        //     }
        // }
        goto(stateIdle)
      }
  }

  //io.pci.AD_OUT := cfgMem.readAsync(U"4'0")
  when(pciFSM.isActive(pciFSM.stateCfgRead)) {
    io.pci.AD_OUT := cfgMem.readAsync(cfgMemAddr)
//  } .elsewhen(pciFSM.isActive(pciFSM.stateMemRead)) {
//    io.pci.AD_OUT := blkMemData
  } .otherwise {
    io.pci.AD_OUT := U"32'0"
  }

  val counter = Reg(UInt(8 bits)) init(0)

  when(io.cond0){
    counter := counter + 1
  }

  io.state := counter
  io.flag  := (counter === 0) | io.cond1
}

//Generate the MyTopLevel's Verilog
object MyTopLevelVerilog {
  def main(args: Array[String]) {
    //SpinalVerilog(InOutWrapper(new MyTopLevel))
    //  SpinalConfig()
    //   .addStandardMemBlackboxing(blackboxAll)
    //   .generateVerilog(new MyTopLevel)
    SpinalVerilog(new MyTopLevel)
  }
}

//Generate the MyTopLevel's VHDL
object MyTopLevelVhdl {
  def main(args: Array[String]) {
    SpinalVhdl(new MyTopLevel)
  }
}

//Define a custom SpinalHDL configuration with synchronous reset instead of the default asynchronous one. This configuration can be resued everywhere
object MySpinalConfig extends SpinalConfig(defaultConfigForClockDomains = ClockDomainConfig(resetKind = SYNC))

//Generate the MyTopLevel's Verilog using the above custom configuration.
object MyTopLevelVerilogWithCustomConfig {
  def main(args: Array[String]) {
    MySpinalConfig.generateVerilog(new MyTopLevel)
  }
}
```

Dolu1990, 2020-04-12:
Hi,

Right, the error is realy weird, but basicaly, the error come because the address you use to read the ram is 10 bits instead of 2 bits. Anyway, i'm looking why the error crash like this, shoudln't have been the case

Dolu1990, 2020-04-12:
wait, the real issue is there :

```scala
object PCIBus {
  val FRAME_ACTIVE = False
  val IDRY_INACTIVE = True
  val IDRY_ACTIVE = False
  val TRDY_INACTIVE = True
  val TRDY_ACTIVE = False
  val IDSEL_ACTIVE = True

  val CMD_MEMORY_READ  = B"0110"
  val CMD_MEMORY_WRITE = B"0111"
  val CMD_CONFIG_READ  = B"1010"
  val CMD_CONFIG_WRITE = B"1011"
}
```

Basicaly, you can't do such things, you shouldn't create netlist stuff outside of the SpinalHDL elaboration, basicaly, here you are defining global signal False. 

Instead of val, use def :)

Dolu1990, 2020-04-12:
So basicaly, the reason why the error message was none-sense, is because the first time it elaborate the design, it figure out that the ram address size was wrong, then, to give a better look at the errors, SpinalHDL redo the elaboration with some specific debug flags set. The issue here is that this time, this re-elaboration of the hardware has the issue that those PCIBus constants are signal part of the first elboration, which trigger those weird errors.

azaparov, 2020-04-12:
Thank you! Still would be great to have little bit more clear error description from the compiler.

Dolu1990, 2020-04-12:
hmm, for that given situation, the best i have right now is : 

```scala
OLD NETLIST RE-USED : (??? :  Bool) is used to drive the (toplevel/io_pci_TRDY : out Bool) := (??? :  Bool) statement, but was defined in another netlist.
Be sure you didn't defined a hardware constant as a 'val' in a global scala object.
    spinal.tester.code.MyTopLevel.<init>(Miaou.scala:70)
    spinal.tester.code.MyTopLevelVerilog$$anonfun$main$1.apply(Miaou.scala:163)
    spinal.tester.code.MyTopLevelVerilog$$anonfun$main$1.apply(Miaou.scala:163)
    spinal.tester.code.MyTopLevelVerilog$.main(Miaou.scala:163)
    spinal.tester.code.MyTopLevelVerilog.main(Miaou.scala)
```

which point to the expression using the val constant.

azaparov, 2020-04-12:
Thank you for your help!

closed 2020-04-27 by Dolu1990