SpinalHDL/SpinalHDL #298: Question about JTAG behavior

yokoyama-flogics, 2020-04-15:
Hello,

Now I'm evaluating JTAG functionality of SpinalHDL (commit: ecb5a80) by the following code.  (Pwm class file is omitted.)  Running on TinyFPGA-BX.

```scala
import spinal.lib.com.jtag.{Jtag, JtagTap}
import flogics.lib.pwm._

class JtagPwm_TinyFPGA_BX extends Component {
  val io = new Bundle {
    val CLK = in Bool
    val PWM = out Bool
    val TCK = in Bool
    val TDI = in Bool
    val TDO = out Bool
    val TMS = in Bool
    val USBPU = out Bool
  }

  val pwm_size = 8

  val core_clock_domain = ClockDomain(
    clock = io.CLK,
    frequency = FixedFrequency(16 MHz),
    config = ClockDomainConfig(
      resetKind = BOOT
    )
  )

  class MyJtagTap extends Component {
    val io = new Bundle {
      val jtag = slave(Jtag())
      val width = out UInt(pwm_size bits)
    }

    val tap = new JtagTap(io.jtag, 8)
    val idcodeArea = tap.idcode(B"x87654321")(instructionId = 4)
    val widthArea = tap.write(io.width)(instructionId = 5)
  }

  val jtag_area = new ClockingArea(ClockDomain(io.TCK)) {
    val jtag = new MyJtagTap()
    jtag.io.jtag.tms <> io.TMS
    jtag.io.jtag.tdi <> io.TDI
    jtag.io.jtag.tdo <> io.TDO
  }

  val core_area = new ClockingArea(core_clock_domain) {
    io.USBPU := False

    val pwm = new Pwm(
      size = pwm_size
    )
    pwm.io.width := jtag_area.jtag.io.width
    io.PWM := pwm.io.output
  }
}

object JtagPwm_TinyFPGA_BX {
  def main(args: Array[String]): Unit = {
    SpinalVerilog(new JtagPwm_TinyFPGA_BX)
  }
}
```

I prepared an FT232H breakout and wrote a code by [PyFtdi](https://pypi.org/project/pyftdi/) as following.

```python
from os import environ
from pyftdi.jtag import JtagEngine, JtagTool
from pyftdi.bits import BitSequence
import sys

url = environ.get('FTDI_DEVICE', 'ftdi://ftdi:232h/1')
jtag = JtagEngine(frequency=100e3)
jtag.configure(url)
jtag.reset()    # Test Logic Reset (5 times of TMS=1)

tool = JtagTool(jtag)

# To Shift-IR, Exit1-IR and finally move to Update-IR
jtag.write_ir(BitSequence('00000100', msb=True, length=8))

# Move to Shift-DR, repeat Shift-DR, wait 15ms, and move to Run-Test/Idle
idcode = jtag.read_dr(32)
print("IDCODE (reset): 0x%x" % int(idcode))

# Move to Run-Test/Idle if required
jtag.go_idle()

# To Shift-IR, Exit1-IR and finally move to Update-IR
jtag.write_ir(BitSequence('00000101', msb=True, length=8))

jtag.write_dr(BitSequence(0, msb=False, length=8))
# The following looks required to reflect DR change!?
jtag.go_idle()

sys.exit(0)
```

The code works fine but I have a question about the **Update-DR** state of JTAG.

Please find the following oscilloscope screen.

- Yellow: TCK
- Light blue: TDI
- Purple: To show when DR (Data Register) is reflected
- Blue: TMS

These signals correspond to the last Shift-DR iteration in the Python code.
(```jtag.write_dr(BitSequence(0, msb=False, length=8))```)

- TDI: 000000000
- TMS: 000000011

At the 9th TCK rising-edge, the JTAG state goes to Update-DR state, I think.

[image]

To my understanding, Data Register (DR) will be updated whenever the JTAG state machine goes to Update-DR state from Exit1-DR.  However, the above chart implies we need to wait for the additional TCK rising edge to reflect DR.

Is this as expected?

Thanks and Regards,
Atsushi Yokoyama

FYI, I'm using Adafruit FT232H Breakout (ID: 2264) old version comes with Micro USB connector.

Dolu1990, 2020-04-17:
Hi,

Yes, basicaly, the effect of things are based on the state, not on the next state, so the write to the register is done when there is a rising edge in the update state, not when going in the update state.

(To, the wave you show is the behaviour i expect)

Then i don't know if the JTAG spec is describing how that should be done, you might be right, and it should be done the way you described

Also, we are currently reworking the JTAG internals a bit : https://github.com/SpinalHDL/SpinalHDL/pull/286

To make it more portable, especialy when you want to integrate some additional instruction into a pre-existing jtag tap, as for example in FPGA.

yokoyama-flogics, 2020-04-18:
Hello,

Thank you for your reply.  I understood it.

I'm also not familiar with JTAG exact specification, and also I don't see practical problem with the current implementation, so it is okay.  (You can close this thread.)

I'm enjoying SpinalHDL and VexRiscv.  I'm looking forward to see the JTAG internal updates!

Meanwhile, I wrote a very small example and published here (GitHub).

- [GitHub: yokoyama-flogics/jtag_pwm_tinyfpga_bx](https://github.com/yokoyama-flogics/jtag_pwm_tinyfpga_bx)

Thanks and Regards,
Atsushi

Dolu1990, 2020-04-18:
Thanks ^^

Didn't know there was such python library to interface FTDI. Cool :D

Regards
Charles

closed 2020-04-18 by Dolu1990