SpinalHDL/SpinalHDL #300: Shift behaviour

saahm, 2020-04-15:
I am getting trouble implementing shift behaviours as available in Verilog. 
In SpinalHDL there is `>>`, `<<` and corresponding `|>>` and `|<<`. According to the documentation these only differ in output size of the resulting bitvector. 
Yet in SpinalHDL `a << b` will translate into `a <<< b` for Verilog. In Verilog `<<<` is arithmetic shift, not logical shift (same issue with shift right). (See https://www.nandland.com/verilog/examples/example-shift-operator-verilog.html for reference of Verilog shift)

The shift operation available in the Int and Bits data types should behave according to the HDL it generates. Luckily VHDL treats the differently with numeric_std, so I assume this issue might only regard Verilog if the generated VHDL makes use of numeric_std.

An example to clarify what I mean
```scala
val a = Bits(32 bits)
val b = Bits(5 bits)
val res = Bits(32 bits)
val resU = UInt(32 bits)

// first with Bits
res := a << b.asUInt // this will throw a width mismatch even tho documentation says output will be Bits(w(x) bits) 
res := a |<< b.asUInt // this will generate a <<< b in verilog for me which is not correct (Bits does not even support arithmetic shifts!)
// with UInt -- which should have difference between arithmetic and logical shifts
resU := a.asUInt << b.asUInt  // this will produce width mismatch (T(w(x) + max(y) bits) being bigger than 32 bits)
resU := a.asUInt |<< b.asUInt  // this will produce a <<< b in verilog again, which is not logical shift!
```
There is no way to make SpinalHDL generate Verilogs `<<` and `>>` or to know before generation if `>>>`/`<<<` will be generated from Spinals `|<<` and other shifts operators.

Dolu1990, 2020-04-15:
Hi,

Is there a difference between << and <<< in verilog ? other than the symbol ? Looking at your link, it seem that << and <<< behave the same. right ?

If it was for >> vs >>> sure, that behave differently.

saahm, 2020-04-16:
Since there is no official available documentation of shift behaviour in Verilog that I could find, I tested it quickly. It has no change to the left shifts, as they seem to behave the same (checked `<<` and `<<<` since that was the question) for signed and unsigned (yet there being different opinions out there to what a logic left shift does compared to an arithmetic left shift). But independent of actual behavior it should generate the proper shift for clarification? I only knew there existed different left shifts too, and I did never dig deeper into it until now.

But then one thing remains: my example doesnt include the right shifts, but it has the same behavior in generated RTL (not creating `>>` but `>>>` in all cases).

Dolu1990, 2020-04-17:
> But independent of actual behavior it should generate the proper shift for clarification

Idealy, yes, basicaly, the way how was implemented the verilog backend was to properly identify how the <<< >>> shifts works, and then limit itself to that behaviour to avoid weird cases.

See : 
https://github.com/SpinalHDL/SpinalHDL/commit/4f7a1dd0fc165b681eecf7f7a783e93240491cd1

In that instance, for some specific version of VCS, the shift behaviour changed XD

So, i agree that things should be done with <<  <<< >>> >>, but unless there is "strong" insentives to update the backend, i prefer keeping it as it is now, instead of taking the risk of breaking something for some given synthesis flow.

Make sense ?

saahm, 2020-04-17:
Yes it makes sense, for now there is no real need to force this update.

closed 2020-04-17 by saahm