SpinalHDL/SpinalHDL #302: One module per file

wswslzp, 2020-04-17:
Hi,

When I simply call the `SpinalVerilog` to generate the verilog file, all the modules defined in the design are all resided in only one file. And I want to find a way to make each module be written into a single file.

I found that there is a parameter named `oneFilePerComponent` defined in the `SpinalConfig` constructor whose default value is false. I set the value to be true and expect that one module in one file, but still all modules remain in the one file. 

Is that function still not be implemented ? If not, how could I do that?

wswslzp, 2020-04-17:
Platform: Windows 10
IDE: IntelliJ IDEA + scala plugin + sbt(1.3.8)
Version: SpinalHDL 1.3.8

Dolu1990, 2020-04-21:
Hi,

Hmm i think that option was something of the past, and is  not implemented for SpinalVerilog.
It would not be realy difficult to bring that feature back to life, but for which purpose ?

>  If not, how could I do that?

Basicaly, each component is added in the main file here (see link bellow), you would just have to deflect that into separate files :)

https://github.com/SpinalHDL/SpinalHDL/blob/dev/core/src/main/scala/spinal/core/internals/PhaseVerilog.scala#L55

saahm, 2020-04-22:
I looked for that feature too but wasnt sure why it wasnt available. 
I can only speak for myself here tho: In the process of working with SpinalHDL i found it difficult to understand what Verilog is gonna be generated. This is essential for some details of behaviour (because some Verilog constructions will have a known and wanted behaviour). So when I design something and want to test it, I start up the SpinalHDL Testbench, yet the ways to interact with the design are different from within a Verilog testbench (slightly, but enough to have a second testbench setup that is written in verilog). Whenever I find an issue in my waves that is unwanted behaviour of my design, i track down what signals and operations are relevant for that. First I can see where in my SpinalHDL i might have the issue. Since I am not fully secure and experienced with the usage of Spinal and its ways to describe behaviour, I might type something that is allowed but does not create what I think it should create. Then I look into the generated Verilog to see what expressions I created. When I see it in Verilog I something think "oh, thats not what I wanted, but I can see why SpinalHDL generated it exactly like this" and can fix it in my SpinalHDL code. Tracking down what happens in my design on Verilog is easier if it isnt a single Verilog file with 1k+ LOC from 5 modules. Even tho I can just ctrl+f in that one file and go by my signal and area names (which not necessarily are preserved all the times).

piessteady, 2020-04-27:
I would also like this feature (to be back) :)
- It can decrease the incremental compile time on big projects
- It can ease debug in large projects -like @saahm is mentioning- the same _zz_1_ is to be found everywhere, most people do not use an IDE that is RTL scope aware.
- Some flows rely on loading a different implementation for a module, some tools only allow to do this by loading in different files for each version.

But I do not know if it is practical, since I guess the number of files generated would be different depended on the small implementation changes.
Maybe a way to tag Components which are "toplevel" modules you would want to see in separate files?

closed 2020-07-20 by wswslzp