SpinalHDL/SpinalHDL #303: elaboration failure upgarding spinal from 1.38 to 1.4.0

aus747, 2020-04-18:
scalaVersion := "2.12.11"

EclipseKeys.withSource := true

libraryDependencies ++= Seq(
  "com.github.spinalhdl" % "spinalhdl-core_2.12" % "1.4.0",
  "com.github.spinalhdl" % "spinalhdl-lib_2.12" % "1.4.0"
)

-----

using spinal 1.3.8 works

this project will reproduce:

[Archive.zip](https://github.com/SpinalHDL/SpinalHDL/files/4496010/Archive.zip)

-----
[Runtime] SpinalHDL v1.4.0    git head : 96467330de902dd3909050e284757af593f5b2d1
[Runtime] JVM max memory : 7282.0MiB
[Runtime] Current date : 2020.04.18 13:38:44
[Progress] at 0.000 : Elaborate components

**********************************************************************************************
[Warning] Elaboration failed (1 error).
          Spinal will restart with scala trace to help you to find the problem.
**********************************************************************************************

[Progress] at 0.230 : Elaborate components
Exception in thread "main" spinal.core.SpinalExit: 

 dslScope stack is not empty :(
********************************************************************************
********************************************************************************

Design's errors are listed above.
SpinalHDL compiler exit stack : 

	at spinal.core.SpinalExit$.apply(Misc.scala:396)
	at spinal.core.SpinalError$.apply(Misc.scala:451)
	at spinal.core.internals.PhaseContext.checkGlobalData(Phase.scala:185)
	at spinal.core.internals.PhaseCreateComponent.impl(Phase.scala:2137)
	at spinal.core.internals.PhaseContext.doPhase(Phase.scala:196)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$10(Phase.scala:2373)
	at spinal.core.internals.SpinalVerilogBoot$.$anonfun$singleShot$10$adapted(Phase.scala:2371)
	at scala.collection.mutable.ResizableArray.foreach(ResizableArray.scala:62)
	at scala.collection.mutable.ResizableArray.foreach$(ResizableArray.scala:55)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:49)
	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2371)
	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2306)
	at spinal.core.Spinal$.apply(Spinal.scala:322)
	at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:148)
	at spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:354)

-----

wswslzp, 2020-04-18:
You should append a sentence to the library dependencies

```scala
libraryDependencies ++= Seq(
  "com.github.spinalhdl" % "spinalhdl-core_2.11" % spinalVersion,
  "com.github.spinalhdl" % "spinalhdl-lib_2.11" % spinalVersion,
// this line is compulsory
  compilerPlugin("com.github.spinalhdl" % "spinalhdl-idsl-plugin_2.11" % spinalVersion)
)
```

HWHaxor, 2020-04-18:
yeah - you're right. that fixed it. should have checked the sample app. thanks.

closed 2020-04-18 by HWHaxor