SpinalHDL/SpinalHDL #306: Verilator compile failure on constant unsigned integer comparison

distributed, 2020-04-22:
I have a component that I could not simulate because Verilator would not compile the Verilog generated by Spinal. I have a reduced repro case.

Luckily I can naturally work around the error as my component does not naturally have the "zero register" of the repro case below. However I believe that the code pattern can show up as a result of some factory/parameter combination that generates a constant register that makes an unsigned integer comparision trivially constant.

The error given by Verilator is 

```
[info] %Warning-UNSIGNED: /home/joe/FPGA/proj/hdl/spinal/tmp/job_1/VerilogComparisonFailure.v:17: Comparison is constant due to unsigned arithmetic
[info] %Warning-UNSIGNED: Use "/* verilator lint_off UNSIGNED */" and lint_on around source to disable this message.
[info] %Error: Exiting due to 1 warning(s)
```

The repro program is:

```scala
import spinal.core.sim._
import spinal.core._
import spinal.sim._
import spinal.lib._
import spinal.lib.fsm._

class VerilogComparisonFailure extends Component {
	val zeroReg = Reg(UInt(8 bits)) init(0)
	val rptr = Reg(UInt(8 bits)) init(0)

	// verilator error attached to this comparison
	val comparisonResult = (rptr < zeroReg)

	// necessary to trigger the error
	rptr := rptr + 1

	// the signal output is not necessary to trigger the described behavior.
	val io = new Bundle {
		val sig = out Bool
	}
	io.sig := comparisonResult
}

object VerilogComparisonFailureSim {
	def main(args: Array[String]) {
		val compiledSim = SimConfig.withWave.compile(new VerilogComparisonFailure())
	}
}
```

The resulting Verilog output

```verilog
// Generator : SpinalHDL v1.4.0    git head : ecb5a80b713566f417ea3ea061f9969e73770a7f
// Date      : 22/04/2020, 15:07:41
// Component : VerilogComparisonFailure

module VerilogComparisonFailure (
  output              io_sig,
  input               clk,
  input               reset 
);
  wire       [7:0]    zeroReg;
  reg        [7:0]    rptr;
  wire                comparisonResult;

  assign zeroReg = 8'h0;
  assign comparisonResult = (rptr < zeroReg);
  assign io_sig = comparisonResult;
  always @ (posedge clk or posedge reset) begin
    if (reset) begin
      rptr <= 8'h0;
    end else begin
      rptr <= (rptr + 8'h01);
    end
  end

endmodule
```

Dolu1990, 2020-04-22:
Hi ^^

Verilator linting <3

So to workaround that just do .addSimulatorFlag("-Wno-UNSIGNED")  on the SimConfig.

I will add that by default in dev :)

closed 2020-04-22 by Dolu1990

distributed, 2020-04-22:
Hi there! I feel the love as well ;) Thanks for the fix.