SpinalHDL/SpinalHDL #308: Current state from StateMachine in datapath logic

saahm, 2020-04-23:
I couldnt find that in the docs and wasnt sure if that is even a possibility. I wanted to access the current state from outside StateMachine in some Area, in order to invoke behaviour based of the state a FSM is in.
For example I have some control logic and some datapath, in the control logic use a FSM via StateMachine. Now when I would create the FSM with a SpinalEnum and all that classical ways of doing that, I have no issue in creating behaviour based of the state. 
It would be good for the design possibilities to still be able to split datapath and controlpath with the usage of StateMachine library.
Any ideas?

Dolu1990, 2020-04-24:
Hi,

Currently, the API solution is to use the myStateMachine.isActive(stateA), which can give you for each state if it's in or not.

But then about about accessing the raw state of the state machine, it is kind of hiden. Is the above enough ?

saahm, 2020-04-24:
Having it in a switch-statement would be more handy, but I will try it that way and see if the resulting hardware results in something similar. 
I will give it a try and get back to this issue. Thanks

saahm, 2020-04-26:
Ok, I have tested it and can confirm some things. Using when-elsewhen-otherwise will generate some hardware using case-statements, so it still creates the behaviour I wanted to have. This would then mean it is infact enough with the current API. The amount of writing is not too bad either once the StateMachine is imported via `import myFSM._` in the Area I want to use it.
Thanks!

closed 2020-04-26 by saahm