SpinalHDL/SpinalHDL #309: Simulation: Assert in StateMachine fails even though it should not

distributed, 2020-04-24:
In many of my state machines, I have asserts inside the `bla = new State { whenIsActive {` block to check for conditions that should hold in this specific state.

Sometimes in simulations, these asserts fail at t=0 with no indication why. The failing assert is contained in states that cannot be reached at t=0 or even states that cannot be reached at all. The obvious exception would be if reset was not asserted, but my understanding is that simulation starts out with reset asserted.

I have reduced the problem to a fairly minimal example. I am going to show the code, the error message and a screenshot of the resultant .vcd file. Some seeds that cause failure (consistently, at least on my machine) are included in the example. In the .vcd, I cannot see where `stateNext` or `stateReg` are ever at `unreachableState`, so everything's shrouded in mystery for me ;)

The code is this:

```scala
import spinal.core.sim._
import spinal.core._
import spinal.sim._
import spinal.lib._
import spinal.lib.fsm._
import scala.util.Random

class BootFail extends Component {
	val io = new Bundle {
		val trig = in Bool
		val sig = out Bool
	}

	io.sig := False
	val sm = new StateMachine {
		val idle: State = new State with EntryPoint {
			whenIsActive {
				when (io.trig) {
					goto(c1)
				}
			}
		}
		val c1: State = new State {
			whenIsActive {
				io.sig := True
				goto(idle)
			}
		}
		val unreachableState: State = new State {
			whenIsActive {
				assert(False, "wound up in unreachableState", FAILURE)
				io.sig := False
				goto(idle)
			}
		}
	}
}

object BootFailSim {
	def main(args: Array[String]) {
		val compiledSim = SimConfig.withWave.compile(new BootFail)

		for (i <- 0 to 1000000) {
			print("running case %d\n".format(i))
			// examples of seeds that lead to failure:
			// 814316022, 1851891454, 533866718, 1197810070
			compiledSim.doSimUntilVoid("case") { dut =>
				dut.clockDomain.forkStimulus(period=10)

				val trigval = i % 2 == 0
				dut.io.trig #= trigval

				dut.clockDomain.waitSampling(10)
				simSuccess()
			}
		}
	}
}
```

When simulating, the error looks like this:
```
[info] [Progress] Start BootFail case simulation with seed 814316022, wave in /home/joe/FPGA/proj/hdl/spinal/./simWorkspace/BootFail/case.vcd
[info] [Error] Simulation failed at time=0
[error] Exception in thread "main" spinal.sim.SimFailure: Verilog assertion failurespinal / Compile / runMain 2s
[error] 	at spinal.sim.SimManager.runWhile(SimManager.scala:204)
[error] 	at spinal.sim.SimManager.runAll(SimManager.scala:190)
[error] 	at spinal.core.sim.SimCompiled.doSimApi(SimBootstraps.scala:219)
[error] 	at spinal.core.sim.SimCompiled.doSimUntilVoid(SimBootstraps.scala:193)
[error] 	at BootFailSim$$anonfun$main$1.apply$mcVI$sp(BootFail.scala:47)
[error] 	at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:160)
[error] 	at BootFailSim$.main(BootFail.scala:43)
[error] 	at BootFailSim.main(BootFail.scala)
[info] FAILURE wound up in unreachableState
[info] - /home/joe/FPGA/proj/hdl/spinal/tmp/job_1/BootFail.v:107: Verilog $finish
[error] Nonzero exit code returned from runner: 1
```

Here's a screenshot of Gtkwave:
[image]

Dolu1990, 2020-04-24:
hmm can't reproduce it in dev branch, i mean, i'm currently at the 70000 retry, and nothing crashed.

So, after how many try it crash for you ?
Which verilator version do you have ?
Which version of SpinalHDL ?

distributed, 2020-04-24:
Sorry, forgot to mention how often I have to run it! Usually, I have to run the loop fewer than 10 times, the highest number I've ever seen was 15.

SpinalHDL is using whatever Verilator is in the PATH, right? So for me that would be:
```
$ /usr/bin/verilator --version
Verilator 3.916 2017-11-25 rev verilator_3_914-65-g0478dbd
```

I am using `spinalVersion = "1.4.0"`.

I will try to reproduce using the dev branch version.

distributed, 2020-04-24:
I built the newest Verilator 4.032 from source and tried to reproduce. I cannot reproduce the failure, with neither Spinal 1.4.0 nor the newest dev branch.

Guess this resolved then. It's a bit of a shame that the latest Ubuntu LTS includes Verilator 3.916 that seems to have the issue :(

Dolu1990, 2020-04-25:
Personnaly, i'm on Verilator 4.032 2020-04-04 rev v4.032
Right, Ubuntu has a realy old version XD

distributed, 2020-04-26:
OK, as we know what to do, I am going to close the issues. Reopen if you don't agree ;)

closed 2020-04-26 by distributed