SpinalHDL/SpinalHDL #316: can I run simulation in notebook (e.g. polynote)?

suda-morris, 2020-05-08:
Hi team!
I'm trying SpinalHDL using polynote (just because it has Scala support by default, don't have to install another kernel again).
I wrote a simple component according to to SpinalHDL project template, I can generate the verilog RTL very well.
[image]
But when it comes to the simulation
```scala
import scala.util.Random

//MyTopLevel's testbench
SimConfig.withWave.doSim(new MyTopLevel){dut =>
  //Fork a process to generate the reset and the clock on the dut
  dut.clockDomain.forkStimulus(period = 10)
  var modelState = 0
  for(idx <- 0 to 99){
    //Drive the dut inputs with random values
    dut.io.cond0 #= Random.nextBoolean()
    dut.io.cond1 #= Random.nextBoolean()
    //Wait a rising edge on the clock
    dut.clockDomain.waitRisingEdge()
    //Check that the dut values match with the reference model ones
    val modelFlag = modelState == 0 || dut.io.cond1.toBoolean
    assert(dut.io.state.toInt == modelState)
    assert(dut.io.flag.toBoolean == modelFlag)
    //Update the reference model value
    if(dut.io.cond0.toBoolean) {
      modelState = (modelState + 1) & 0xFF
    }
  }
}
```
 I got the following issue:
```bash
[Runtime] SpinalHDL v1.3.8    git head : 57d97088b91271a094cebad32ed86479199955df
[Runtime] JVM max memory : 1744.0MiB
[Runtime] Current date : 2020.05.08 10:28:22
[Progress] at 7.114 : Elaborate components
[Progress] at 7.129 : Checks and transforms
[Progress] at 7.181 : Generate Verilog
[Done] at 7.186
[Progress] Simulation workspace in /opt/polynote/./simWorkspace/MyTopLevel
[Progress] Verilator compilation started
Line Number->2
code->compiler.err.doesnt.exist
Message->package spinal.sim does not exist
Source->spinal.sim.DynamicCompiler$InMemoryJavaFileObject[string:///wrapper_verilator/VerilatorNative.java]
 
Line Number->4
code->compiler.err.cant.resolve
Message->cannot find symbol
  symbol: class IVerilatorNative
Source->spinal.sim.DynamicCompiler$InMemoryJavaFileObject[string:///wrapper_verilator/VerilatorNative.java]
```
```bash
    Uncaught exception: (java.lang.NullPointerException)

        spinal.sim.VerilatorBackend.<init>(VerilatorBackend.scala:433)
        spinal.core.sim.SpinalVerilatorBackend$.apply(SimBootstraps.scala:120)
        spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:400)
        spinal.core.sim.SpinalSimConfig.compile(SimBootstraps.scala:364)
        spinal.core.sim.SpinalSimConfig.doSim(SimBootstraps.scala:350)
        (Line 4)
        sun.reflect.NativeConstructorAccessorImpl.newInstance0(NativeConstructorAccessorImpl.java:-2)
        sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        polynote.kernel.interpreter.scal.ScalaInterpreter.polynote$kernel$interpreter$scal$ScalaInterpreter$$createInstance(ScalaInterpreter.scala:157)
        polynote.kernel.interpreter.scal.ScalaInterpreter$$anonfun$polynote$kernel$interpreter$scal$ScalaInterpreter$$runClass$3$$anonfun$apply$24.apply(ScalaInterpreter.scala:169)
        zio.blocking.package$Blocking$Service$$anonfun$effectBlockingInterrupt$1$$anonfun$apply$3$$anonfun$apply$4.apply(package.scala:126)
        zio.blocking.package$Blocking$Service$$anonfun$effectBlockingInterrupt$1$$anonfun$apply$3$$anonfun$apply$4.apply(package.scala:118)
        zio.ZIO$$anonfun$effectSuspend$1.apply(ZIO.scala:2251)
        zio.ZIO$$anonfun$effectSuspend$1.apply(ZIO.scala:2251)
        zio.internal.FiberContext.liftedTree1$1(FiberContext.scala:539)
        zio.internal.FiberContext.evaluateNow(FiberContext.scala:539)
        zio.internal.FiberContext.zio$internal$FiberContext$$run$body$1(FiberContext.scala:681)
        zio.internal.FiberContext$$anonfun$11.run(FiberContext.scala:681)
        polynote.kernel.interpreter.CellExecutor$$anon$3$$anonfun$run$1$$anonfun$apply$mcV$sp$1.apply$mcV$sp(CellExecutor.scala:33)
        polynote.kernel.interpreter.CellExecutor$$anon$3$$anonfun$run$1$$anonfun$apply$mcV$sp$1.apply(CellExecutor.scala:33)
        polynote.kernel.interpreter.CellExecutor$$anon$3$$anonfun$run$1$$anonfun$apply$mcV$sp$1.apply(CellExecutor.scala:33)
        scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        scala.Console$.withOut(Console.scala:53)
        polynote.kernel.interpreter.CellExecutor$$anon$3$$anonfun$run$1.apply$mcV$sp(CellExecutor.scala:32)
        polynote.kernel.interpreter.CellExecutor$$anon$3$$anonfun$run$1.apply(CellExecutor.scala:31)
        polynote.kernel.interpreter.CellExecutor$$anon$3$$anonfun$run$1.apply(CellExecutor.scala:31)
        polynote.kernel.package$.withContextClassLoader(package.scala:99)
        polynote.kernel.interpreter.CellExecutor$$anon$3.run(CellExecutor.scala:30)
        java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        java.lang.Thread.run(Thread.java:748)
```
BTW, I'm using SpinalHDL version: com.github.spinalhdl:spinalhdl-core_2.11:1.3.8 and com.github.spinalhdl:spinalhdl-lib_2.11:1.3.8

c-thaler, 2020-05-08:
SpinalHDL simulation uses verilator as backend. Have you build and installed verilator on your system?

suda-morris, 2020-05-09:
> SpinalHDL simulation uses verilator as backend. Have you build and installed verilator on your system?

Thanks for replying.
Yes, I installed verilator using pacman in my system.
```bash
➤ verilator --version                                                                                                              10:29:14
Verilator 4.030 2020-03-08 rev v4.028-50-g95c4b6aa
```

gyn, 2020-05-09:
> > SpinalHDL simulation uses verilator as backend. Have you build and installed verilator on your system?
> 
> Thanks for replying.
> Yes, I installed verilator using pacman in my system.
> 
> ```shell
> ➤ verilator --version                                                                                                              10:29:14
> Verilator 4.030 2020-03-08 rev v4.028-50-g95c4b6aa
> ```

Did you check if you could execute programs like ls or ps in your PATH inside polynote?

suda-morris, 2020-05-09:
Thank you @gyn 

I checked it in this way, seems like polynote can execute these commands.

[image]

typingArtist, 2020-06-20:
Issue confirmed here while upgrading SpinalWorkshop sbt tests to Mill. Don’t know yet why and exactly when it happens to me. Will follow-up with that.

numero-744, 2022-08-18:
Hello, any updates about this?

closed 2022-11-16 by numero-744